enjoy-digital/litex #577: CSRStorage not reset on soft reboot

madscientist159, 2020-06-25:
I have noticed that any CSRStorage objects declared with a reset= value (e.g. `self.config = CSRStorage(size=32, reset=0x12345678`) will only be reset to that value on initial FPGA configuration.  Running a soft reset of the CPU (`reboot`) leaves the CSRStorage registers at the prior values from before the reboot.

This has caused some degree of problems when e.g. a new peripheral core is now coming up in an unknown state and the bootloader can't handle configuration from that state.  Is there a known / preferred method to ensure the CSRStorage registers are reset on soft CPU reboot?

enjoy-digital, 2020-06-27:
The `reboot` command of the BIOS is indeed currently only reseting the CPU and not the peripherals. We could change this behavior, in the meantime, the `soc.ctrl.reset`could be connected the `CRG` to reset the full SoC.

madscientist159, 2020-06-27:
@enjoy-digital What is the correct method to attach `soc.ctrl.reset` to the `CRG`?

enjoy-digital, 2020-07-01:
You could add a `reset` signal to the CRG and do in the SoC : `self.comb += self.crg.reset.eq(self.ctrl.reset)`.

enjoy-digital, 2020-11-04:
The` reboot` command is now doing a full reset of the SoC with: https://github.com/enjoy-digital/litex/commit/ffc554deded2990c89dc66f12ce8e2755a7414fe and https://github.com/litex-hub/litex-boards/commit/2b17dc1b89894facc34495f783d670df351fd090.

closed 2020-11-04 by enjoy-digital