enjoy-digital/litex #586: Regression tests For BlackParrot to identify broken commits

scanakci, 2020-07-10:
Hi @enjoy-digital , @mithro 

Thanks for your help in integrating BlackParrot into LiteX!
Our development of BlackParrot is coming along well and we are now booting Linux in LiteX. However, since both BlackParrot and LiteX are evolving, a major issue has been tracking down the root cause of failures when either BP or LiteX are updated (e.g., https://github.com/enjoy-digital/litex/issues/493.) As part of our CI, we (@taylor-bsg, @ajayjoshi) are working on a mechanism on the BP side to run a regression on LiteX to verify that the new BP works. But we could really use the same kind of CI for LiteX, for both simulation and FPGA (as they often diverge), that would automatically flag if a LiteX pull request breaks the processor builds (e.g., VexRISC and BlackParrot).
I wrote two initial test cases in this repo (https://github.com/scanakci/linux-on-litex-blackparrot) which basically fetches LiteX and checks if BlackParrot can run BIOS. Could we get this integrated into LiteX?

Regards,
Sadullah

enjoy-digital, 2020-07-10:
Hi @scanakci,

thanks for your work on BlackParrot integration and sorry for the delay to investigate on #493, i'll work on that in the next days. For others projects built with LiteX, we are running simulation with verilator for CI and enabling it on LiteX for the different CPU would also indeed be a good idea. I'll also try to work on this.

Regards,

Florent

scanakci, 2020-07-10:
Hi @enjoy-digital,

Thank you for your help. I want to clarify one issue, though. You already fixed the #493. I did not create an issue for the new problem that I am facing (related to memtest failure). I sent the details of new problem as an email.

Thanks,

Sadullah

mithro, 2020-07-10:
FYI - The team at Antmicro has also been looking into improving the CI situation. That includes people like @kgugala, @mateusz-holenko and @PiotrZierhoffer.

enjoy-digital, 2021-06-17:
Closing, this will be covered by #763.

closed 2021-06-17 by enjoy-digital