enjoy-digital/litex #589: ECP5 bitstream problem with --spimode

derekmulcahy, 2020-07-13:
The recent PR #588  to add spimode as a parameter breaks the bitstreams for my color light 5A-75E when programmed using an FT2232 and openFPGALoader. The bitstreams being produced by ecppack when the --spimode flag is specified cannot be programmed by openFPGALoader. Here is the very trimmed down sequence of events.

```
build$ ecppack top.config --svf top.svf --bit top.bit --bootaddr 0 --spimode fast-read
build$ openFPGALoader -c ft2232 top.bit
Open file top.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Fail
displayReadReg
        Config Target Selection : 0
        JTAG Active
        ISC Enable
        Write Enable
        Read Enable
        Abort ERR
        EXEC Error
build$ ecppack top.config --svf top.svf --bit top.bit --bootaddr 0
build$ openFPGALoader -c ft2232 top.bit
Open file top.bit DONE
Parse file DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: [==================================================] 100.000000%
Done
Disable configuration: DONE

```

The second run of ecppack, without the --spimode flag, produces a bitstream which can be programmed. The first run is consistently broken.

I realize that this is almost certainly an ecppack issue but the changes here to use --spimode triggered it for me so maybe someone else sees it and has an idea. I will submit an issue for ecppack.

derekmulcahy, 2020-07-13:
Ecppack puts "0x79 0x49 0x00 0x00" into the bitstream when --spimode fast-read is set.  dual-spi spimode adds "0x79 0x51 0x00 0x00" and qspi adds "0x79 0x59 0x00 0x00".

When the --spimode flag is omitted then this sequence is omitted completely.

My 5A-75E flash chip doesn't like any of the above spimode options so I would need to be able to omit the spimode flag. Perhaps it could default to nothing instead of fast-read. It begins to look like it isn't really an ecppack issue after all.

derekmulcahy, 2020-07-13:
OK, a last comment. If I use

`openFPGALoader -c ft2232 -f top.bit`

to write the bitstream to flash then the bitstream produced by --spimode fast-read works! My flash chip does like the fast-read mode after all.

I guess the bottom line is that bitstreams created with spimode options cannot be used for SRAM uploads.

daveshah1, 2020-07-13:
Yes, that is correct, the SPI mode flags break uploads via JTAG or other SRAM programming routes. prjtrellis actually strips them automagically when creating an svf file, but that doesn't help when creating a bit file for another programming tool.

enjoy-digital, 2020-07-13:
Thanks @derekmulcahy, @daveshah1, with https://github.com/enjoy-digital/litex/commit/6671eb6218ce35a4b79f3c6dbbb2092bcaa3b41f spimode is set to None by default (and --spimode is not passed to ecppack in this case).

closed 2020-07-13 by enjoy-digital

derekmulcahy, 2020-07-13:
Awesome! Fixed while I slept!