enjoy-digital/litex #590: Add AXI Lite components/interconnect equivalent to the existing Wishbone ones

enjoy-digital, 2020-07-13:
The main bus of the SoC is for now limited to Wishbone which is simple and easy but starts being a bottleneck on several systems: 
- Most of the newly added CPU (Zynq, BlackParrot, Rocket, etc...) use native AXI interfaces and the bridges to Wishbone are limiting efficiency/performance.
- Most of the cores are now supporting AXI which has become the de facto standard for performance and interoperability (Ours like LiteDRAM, LitePCIe, etc... and external cores),

Supporting an AXI-Lite bus as the main bus of the SoC would improve the overall performance and interoperability (as described in more details in https://github.com/enjoy-digital/litex/issues/372) and would be a good first step before supporting AXI-Full.

With the current AXI definition and modules in https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/axi.py we are already able make the SoC interoperate with internal/external AXI components while still using Wishbone as the main bus.

We should progressively add the AXI-Lite components/interconnect equivalent to the existing Wishbone ones:
- [ ] AXILiteSRAM.
- [ ] AXILite2CSR.
- [ ] AXILiteConverter.
- [ ] AXILiteInterconnectPointToPoint.
- [ ] AXILiteArbiter.
- [ ] AXILiteDecoder.
- [ ] AXILiteInterconnectShared.
- [ ] AXILiteCrossbar.

In a first time, the components that could be easily tested on real SoCs while still using a Wishbone bus as the main bus should be added: AXILiteSRAM, AXILite2CSR, AXILiteConverter. To allow them to be used in a SoC,  the `add_adapter` function of the `SoCBusHandler` would need to be adapted to enable bridges between AXILite <--> Wishbone using the existing bridging components.

Once these components have been validated, the interconnect should be created and tested with unit-tests, then integrated in the SoC Builder and tested on a simple SoC: CPU + ROM + SRAM + CSRBridge + UART and then on real-world designs where the only change from the user would be to specify the `bus_standard` to "axi-lite".

xobs, 2020-07-13:
How difficult would it be to add bridge equivalents to all of these? I'd like for the existing `wishbone-bridge` and `wishbone-tool` stuff to continue to work, and it would be nice if the protocol would be similar.

enjoy-digital, 2020-07-13:
@xobs: the idea is to make it transparent to the user (and do not break anything). Even if the main bus is an AXI-Lite bus, you will still be able to plug a Wishbone component to it (an adapter will be automatically inserted, as we are already doing for example for data width converters https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L281 but here the bus standard would be adapted). So we don't really need to create bridge equivalents, since the bridge are probably not limited by the Wishbone bus (too slow to take advantage of an AXI bus) and even if we create AXI equivalents, we should indeed keep the protocol similar, so the big change you'll have to do is to eventually replace wishbone in the name of the tools :)

closed 2020-07-22 by enjoy-digital

enjoy-digital, 2020-07-23:
@jedrzejboczar: i just did a quick comparison on a VexRiscv SMP design on Arty using a Wishbone bus or AXI-Lite bus:

Both are booting Linux with in a very similar boot time:

Wishbone version:
````

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jul 23 2020 10:29:57
 BIOS CRC passed (0378c6f1)

 Migen git sha1: 7bc4eb1
 LiteX git sha1: 8bdf6941

--=============== SoC ==================--
CPU:       VexRiscv SMP-DEFAULT @ 125MHz
BUS:       WISHBONE 32-bit @ 4GiB
CSR:       8-bit data
ROM:       64KiB
SRAM:      8KiB
MAIN-RAM:  262144KiB

--========== Initialization ============--
Ethernet init...
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |10000000000000000000000000000000| delays: 00+-00
m0, b09: |00011111111111000000000000000000| delays: 08+-06
m0, b10: |00000000000000011111111111000000| delays: 20+-05
m0, b11: |00000000000000000000000000011111| delays: 29+-02
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b09 delays: 08+-06
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |11000000000000000000000000000000| delays: 01+-01
m1, b09: |00011111111111000000000000000000| delays: 08+-06
m1, b10: |00000000000000011111111111100000| delays: 21+-06
m1, b11: |00000000000000000000000000001111| delays: 30+-02
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b10 delays: 21+-06
[... 308 lines cut ...]
[    1.954077] i2c /dev entries driver
[    1.985088] NET: Registered protocol family 10
[    2.001676] Segment Routing with IPv6
[    2.005642] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    2.032926] Freeing unused kernel memory: 180K
[    2.037187] This architecture does not have kernel memory protection.
[    2.045324] Run /init as init process
mount: mounting tmpfs on /dev/shm failed: Invalid argument
mount: mounting tmpfs on /tmp failed: Invalid argument
mount: mounting tmpfs on /run failed: Invalid argument
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Saving random seed: [    3.350787] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting network: OK

Welcome to Buildroot
buildroot login: 
````

And resource usage is also very similar (AXI-Lite vs Wishbone):
[image]

So it seems to be already usable.