enjoy-digital/litex #596: Command aliases are missing following installation instructions

ghent360, 2020-07-19:
Downloaded litex_setup.py and run
./litex_setup.py init install --user

Then
./litex_setup.py gcc

lxsim command is not available, neither is lxterm (actually lxterm is a different binary in ubuntu). I figured these are aliases to python scripts in litex/litex/tools/ Is there something missing from the install process? these should be installed as aliases in the shell somehow.

mithro, 2020-07-19:
@ghent360 Did the output complain about not having `~/.local/bin` not in your `$PATH`?

ghent360, 2020-07-19:
Perhaps it did, but I have not noticed it. I would politely suggest you add a reminder in the README.md instructions.
Also in ubuntu and probably other distros there is /usr/bin/lxterm binary, which is a wrapper around xterm, perhaps renaming the alias would be warranted.

ghent360, 2020-07-19:
@mithro Thank you for the help.

closed 2020-07-19 by ghent360