enjoy-digital/litex #601: no platform file for SoCZynq

rlipperts, 2020-07-23:
Upon discovery of the [SoCZynq SoC](https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_zynq.py) I tried to find working configurations using it but couldn't find any.

- I tried building a platform file for the ZyboZ7 myself (created from the information of the [xdc file](https://github.com/Digilent/digilent-xdc/blob/master/Zybo-Z7-Master.xdc)) but encountered errors because of missing ps7_<...> attributes
```AttributeError: 'BaseSoC' object has no attribute 'ps7_name'```
- As the SoCZynq SoC specifically requests them from the platform they seem to be included in a platform file I could not find

Can anyone provide me with a platform file designed to work with the SoCZynq SoC? If not, are there working examples for the ZyboZ7 that I missed?

enjoy-digital, 2020-07-23:
@rlipperts: thanks for the feedback. I was planning to update the Zynq support, so will do it and provide an example for the Zybo.

enjoy-digital, 2020-07-23:
@rlipperts: the Zynq integration has been updated with https://github.com/enjoy-digital/litex/commit/041c7527ce1b378765c410e6d6e26e4883ee5e44. The SoCs created on Zynq are now regular SoCs and you can choose to use 
- no CPU  :`--cpu-type=None`, 
- a softcore CPU (default equivalent to `--cpu-type=vexriscv`)
- the PS7 of the Zynq with (default equivalent to `--cpu-type=zynq7000`) and then provide the `.xci` you generated with Vivado to the design with `self.set_ps7_xci method`. 

An example for the Zybo Z7 has been added to LiteX-Boards with https://github.com/litex-hub/litex-boards/commit/8a3b453e2f27ebcfe25c7217503197284a8a77e0 and https://github.com/litex-hub/litex-boards/commit/1e1589a5142cedd3bd37ef65bd81cfcf37918e5e.

`./zybo_z7.py --cpu-type=vexriscv --build` with build a SoC with a VexRiscv SoftCore CPU.
`./zybo_z7.py --cpu-type=zynq7000 --build` with build a SoC with the PS7 (but will not build the BIOS, only the headers for the SoC that you could use in your software to access the peripherals).

The Zynq support is still minimal, but it should allow you to get started.

derekmulcahy, 2020-07-23:
> @rlipperts: the Zynq integration has been updated with 041c752. The SoCs created on Zynq are now regular SoCs and you can choose to use
> An example for the Zybo Z7 has been added to LiteX-Boards with litex-hub/litex-boards@8a3b453 and litex-hub/litex-boards@1e1589a.

This sounds awesome. Is it time to resurrect my snickerdoodle?

enjoy-digital, 2020-07-23:
@derekmulcahy: at least I also have a SnickerDoodle so could help if you want to play with this.

enjoy-digital, 2020-07-28:
Closing since the initial issue/question has been solved/answered.

closed 2020-07-28 by enjoy-digital

rlipperts, 2020-07-28:
Sorry for the late reply, I only have a very limited amount of time to work on this.
I just tested it and it worked flawlessly (with slight changes for my zybo 7z020). 
Thank you very much!