enjoy-digital/litex #608: vexriscv_smp breaks designs that don't use LiteDRAM

bunnie, 2020-07-28:
The header-level module dependency pulled into vexriscv_smp means that designs that don't use litedram and also don't use vexriscv_smp now break unless they also have litedram as an installed dependency.

https://github.com/enjoy-digital/litex/blob/59b95fad9c7e595116c8f80a3ce96a6fa11bfbb7/litex/soc/cores/cpu/vexriscv_smp/core.py#L15

See error on Betrusted CI builder: https://ci.betrusted.io/job/soc-agnostic/510/console

It would be nice if we could follow what's been done here:

https://github.com/enjoy-digital/litex/blob/59b95fad9c7e595116c8f80a3ce96a6fa11bfbb7/litex/soc/integration/soc.py#L1114

where litedram is only pulled in as a dependency if DRAM is explicitly involved in a design.

thanks!

enjoy-digital, 2020-07-28:
Thanks for reporting @bunnie, i indeed received a notification things were broken on your CI and was going to look at it but you've been faster :). This should be solved with https://github.com/enjoy-digital/litex/commit/fe38e12b2190253b784fe9c9301ce10773c3b1b3.

bunnie, 2020-07-28:
looks good. thank you!

closed 2020-07-28 by bunnie