enjoy-digital/litex #609: Arty A7 SoC with Lite+Debug VexRiscV does not complete memory test

alanvgreen, 2020-07-29:
I have a Lite+Debug VexRiscV based SoC on an Arty that is failing to get past the memory test.

GDB says:
```
(gdb) step
Note: CPU is currently in a trap: Illegal instruction 0x0000500f at 0x00004098
37	  lw x6,  13*4(sp)
(gdb) 
```
Instruction x500f is dcache flush.

Things that make this work:
*    Works with cpu_variant 'full+debug'
*    Works if I revert 408d1a9f "cpu/vexriscv/system.h: update flush_cpu_dcache." 

Here is the failing boot sequence, up to the point where it hangs.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jul 29 2020 14:40:47
 BIOS CRC passed (b3fe279b)

 Migen git sha1: 7bc4eb1
 LiteX git sha1: 9518ccf

--=============== SoC ==================--
CPU:       VexRiscv @ 100MHz
BUS:       WISHBONE 32-bit @ 4GiB
CSR:       8-bit data
ROM:       32KiB
SRAM:      8KiB
L2:        8KiB
MAIN-RAM:  262144KiB

--========== Initialization ============--
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |11111111111100000000000000000000| delays: 06+-06
m0, b09: |00000000000011111111111111100000| delays: 20+-08
m0, b10: |00000000000000000000000000000111| delays: 30+-01
m0, b11: |00000000000000000000000000000000| delays: -
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b09 delays: 20+-08
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |11111111111100000000000000000000| delays: 06+-06
m1, b09: |00000000000001111111111111110000| delays: 20+-07
m1, b10: |00000000000000000000000000000111| delays: 30+-01
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b09 delays: 20+-07
SDRAM now under hardware control
Memtest at 0x40000000...
```

enjoy-digital, 2020-07-29:
Thanks for reporting @alanvgreen, i confirm i can reproduce the issue with `lxsim --cpu-type=vexriscv --cpu-variant=lite --with-sdram` and will look at it.

enjoy-digital, 2020-07-31:
This is fixed with https://github.com/enjoy-digital/litex/commit/e0a763e534ccbe8689c62906f59f5f9ebfbc27ed, the issue was that the Lite version does not have a data cache and does not support the dcache flush instruction. We are now providing empty `flush_icache`/`flush_dcache` functions for variants whithout i/d caches.

closed 2020-07-31 by enjoy-digital