enjoy-digital/litex #616: undefined reference to `_irq_mask' in PicoRV32 on ULX3S

pepijndevos, 2020-08-01:
When I run the following command:
```
python ulx3s.py --cpu-type picorv32 --cpu-variant minimal --device LFE5U-85F --sdram-module AS4C32M16 --csr-svd ulx3s.svd --build --load
```
I get linking errors:
```
riscv64-unknown-elf-ld: isr.o: in function `isr':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:60: undefined reference to `_irq_pending'
riscv64-unknown-elf-ld: isr.o: in function `isr':
/home/pepijn/code/litex_workspace/litex/litex/soc/software/bios/isr.c:105: undefined reference to `_irq_pending'
riscv64-unknown-elf-ld: boot.o: in function `get_uint32':
/home/pepijn/code/litex_workspace/litex/litex/soc/software/bios/boot.c:139: undefined reference to `_irq_setmask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/software/bios/boot.c:141: undefined reference to `_irq_disable'
riscv64-unknown-elf-ld: main.o: in function `irq_setmask':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:55: undefined reference to `_irq_setmask'
riscv64-unknown-elf-ld: main.o: in function `irq_setie':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:39: undefined reference to `_irq_enable'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_getie':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:33: undefined reference to `_irq_enabled'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `uart_read':
/home/pepijn/code/litex_workspace/litex/litex/soc/software/libbase/uart.c:59: undefined reference to `_irq_enabled'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_getie':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:33: undefined reference to `_irq_enabled'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `uart_write':
/home/pepijn/code/litex_workspace/litex/litex/soc/software/libbase/uart.c:80: undefined reference to `_irq_enabled'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_getmask':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_setmask':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:55: undefined reference to `_irq_setmask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:55: undefined reference to `_irq_setmask'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_getmask':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: /home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:48: undefined reference to `_irq_mask'
riscv64-unknown-elf-ld: ../libbase/libbase-nofloat.a(uart.o): in function `irq_setmask':
/home/pepijn/code/litex_workspace/litex/litex/soc/cores/cpu/picorv32/irq.h:55: undefined reference to `_irq_setmask'
```
It seems these are externed in https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/picorv32/irq.h and defined in https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/picorv32/crt0.S
But apparently these are not linked correctly?

enjoy-digital, 2020-08-01:
Could you try with a clean build directory? `rm -rf build` before the command. It's probably that you compiled code for VexRiscv before which is the default CPU.

pepijndevos, 2020-08-01:
Never mind me.... *facepalm*

closed 2020-08-01 by pepijndevos