enjoy-digital/litex #629: PermissionError: [Errno 13] Permission denied: 'nexys4ddr.v'

0ena, 2020-08-12:
Hi,

I am trying to create a bitstream for a Nexus 4 board using this tutorial:
https://insights.sei.cmu.edu/sei_blog/2019/10/how-to-build-a-trustworthy-freelibre-linux-capable-64-bit-risc-v-computer.html

Running the command `litex/litex/boards/targets/nexys4ddr.py   --gateware-dir /eda/tools/xilinx/Vivado/2019.2   --csr-csv ./csr_nexys4ddr.csv   --with-ethernet --sys-clk-freq=75e6  --cpu-type rocket --cpu-variant linux` I get the following errors:

```
....
 CC       cmd_mem.o
 CC       cmd_boot.o
 CC       cmd_i2c.o
 CC       cmd_spiflash.o
 CC       cmd_litedram.o
 CC       cmd_liteeth.o
 CC       cmd_litesdcard.o
 CC       main.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/user1/Proj/riscv/rocket/slLitex/build/nexys4ddr/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 27.70KiB 	(86.55%)
RAM usage: 1.77KiB 	(22.07%)

make: Leaving directory '/home/user1/Proj/riscv/rocket/slLitex/build/nexys4ddr/software/bios'
Traceback (most recent call last):
  File "litex/litex/boards/targets/nexys4ddr.py", line 123, in <module>
    main()
  File "litex/litex/boards/targets/nexys4ddr.py", line 116, in main
    builder.build(run=args.build)
  File "/home/user1/Proj/riscv/rocket/slLitex/litex/litex/soc/integration/builder.py", line 211, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/user1/Proj/riscv/rocket/slLitex/litex/litex/soc/integration/soc.py", line 1040, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/user1/Proj/riscv/rocket/slLitex/litex/litex/build/xilinx/platform.py", line 50, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/user1/Proj/riscv/rocket/slLitex/litex/litex/build/xilinx/vivado.py", line 322, in build
    v_output.write(v_file)
  File "/home/user1/.local/lib/python3.6/site-packages/migen-0.9.2-py3.6.egg/migen/fhdl/conv_output.py", line 31, in write
    with open(main_filename, "w") as f:
PermissionError: [Errno 13] Permission denied: 'nexys4ddr.v'
```

I am really really stuck with this one. Any ideas on why this might be happening?
Thank you in advance for your responses!

Kind regards,
Nassos

p.s. I tried to run it as su, but things where worse as it couldn't even start, giving this error:

```
Traceback (most recent call last):
  File "litex/litex/boards/targets/nexys4ddr.py", line 9, in <module>
    from migen import *
ModuleNotFoundError: No module named 'migen'
```
But I think running it as su, is not the right way to solve this issue.

cklarhorst, 2020-08-13:
I have not used that target but I think that  `--gateware-dir ` in your command needs to be writable by your user. That is the directory where the verilog & synthesis files will be created. Likely you could just omit that parameter.

gsomlo, 2020-08-13:
@0ena : you probably meant to say `--toolchain-dir` rather than `--gateware-dir` (indeed, as @cklarhorst pointed out, the latter needs to be writable by you, and you're probably better off going with the implicit default).

The reason running as root won't work is that all of the LiteX python modules were installed as your non-root user, so they're likely not available system-wide, to any users besides the (non-root) one you used to install them.

Finally, may I suggest going with [this link](http://www.contrib.andrew.cmu.edu/~somlo/BTCP) instead, as it's the one I have write access to, and can keep (more) updated than the one you cited :) (note to self, it might *really* be time to set up a linux-on-litex-rocket repo on litex-hub and keep all up-to-date build instructions there...)

0ena, 2020-08-17:
Hi @cklarhorst and @gsomlo !
Thank you for your informative posts, they are really appreciated. Sorry for being quiet, but I am doing a lot in parallel. :)

So I tried the `--toolchain-dir` option but I get the same thing as with the `--gateware-toolchain-path` option, namely:

```
litex/litex/boards/targets/nexys4ddr.py --toolchain-dir /eda/tools/xilinx/Vivado/2019.2 --csr-csv ./csr_nexys4ddr.csv --with-ethernet --sys-clk-freq=75e6 --cpu-type rocket --cpu-variant linux
usage: nexys4ddr.py [-h] [--build] [--load] [--output-dir OUTPUT_DIR]
                    [--gateware-dir GATEWARE_DIR]
                    [--software-dir SOFTWARE_DIR] [--include-dir INCLUDE_DIR]
                    [--generated-dir GENERATED_DIR] [--no-compile-software]
                    [--no-compile-gateware] [--csr-csv CSR_CSV]
                    [--csr-json CSR_JSON] [--csr-svd CSR_SVD]
                    [--memory-x MEMORY_X] [--bus-standard BUS_STANDARD]
                    [--bus-data-width BUS_DATA_WIDTH]
                    [--bus-address-width BUS_ADDRESS_WIDTH]
                    [--bus-timeout BUS_TIMEOUT] [--cpu-type CPU_TYPE]
                    [--cpu-variant CPU_VARIANT]
                    [--cpu-reset-address CPU_RESET_ADDRESS]
                    [--integrated-rom-size INTEGRATED_ROM_SIZE]
                    [--integrated-rom-file INTEGRATED_ROM_FILE]
                    [--integrated-sram-size INTEGRATED_SRAM_SIZE]
                    [--integrated-main-ram-size INTEGRATED_MAIN_RAM_SIZE]
                    [--csr-data-width CSR_DATA_WIDTH]
                    [--csr-address-width CSR_ADDRESS_WIDTH]
                    [--csr-paging CSR_PAGING] [--csr-ordering CSR_ORDERING]
                    [--ident IDENT] [--ident-version IDENT_VERSION]
                    [--no-uart] [--uart-name UART_NAME]
                    [--uart-baudrate UART_BAUDRATE]
                    [--uart-fifo-depth UART_FIFO_DEPTH] [--no-timer]
                    [--no-ctrl] [--l2-size L2_SIZE]
                    [--min-l2-data-width MIN_L2_DATA_WIDTH]
                    [--max-sdram-size MAX_SDRAM_SIZE]
                    [--sys-clk-freq SYS_CLK_FREQ] [--with-ethernet]
                    [--with-spi-sdcard] [--with-sdcard]
nexys4ddr.py: error: unrecognized arguments: --toolchain-dir /eda/tools/xilinx/Vivado/2019.2

```

It seems that those two options do not exist anymore?

On the other hand I took a look on the link you provided (http://www.contrib.andrew.cmu.edu/~somlo/BTCP/) and I saw that the command there is formulated with the `--build` option, omitting completely the toolchain directory. My command now is:

`litex/litex/boards/targets/nexys4ddr.py --build --csr-csv ./csr_nexys4ddr.csv --with-ethernet --sys-clk-freq=75e6 --cpu-type rocket --cpu-variant linux`

It worked without giving any kind of error and it produced the bitstream.
@gsomlo I will be more than happy to use your provided link and post any issues I may come up with, in order for you to keep it updated.

You can close this thread now, thank you again folks!

closed 2020-08-19 by gsomlo