enjoy-digital/litex #630: Wishbone no longer meets timing

xobs, 2020-08-14:
On Fomu's CI, today's build no longer meets timing.

The build -- available at https://ci.betrusted.io/job/foboot/165/ -- runs much longer than it used to. It used to take just under three minutes, but now it's taking [four hours](https://ci.betrusted.io/job/foboot/buildTimeTrend).

Additionally, it no longer meets timing, and thus fails to build.

enjoy-digital, 2020-08-14:
Thanks for reporting, this was probably introduced by https://github.com/enjoy-digital/litex/commit/e4f5dd987eb8d2a98c714d4e2130fe015e1df244 (to fix https://github.com/litex-hub/linux-on-litex-vexriscv/issues/149) and should be fixed with with https://github.com/enjoy-digital/litex/commit/35929c0f8a8f1cc098a6b6ebb569caca8df8c08d that reverts the non-registered CSR bridge for SoCs without SDRAM. The issue with Foboot is that the FPGA before this change was almost full (97% LUTs) and registering the CSR bridge is enough to make it fails. I'm aware lot's of functionalities have been packed in Foboot but i would not recommend using an FPGA with a resource usage that high since there will always be small variations in upstream version of the tools (in the SoC builder, the synthesis, place and route tools) and some margins are needed. At least it's a good CI for resource efficiency, but i wouldn't mind having more margin :)

xobs, 2020-08-15:
Thanks! That does fix it.

I'm holding the synthesis and place-and-route tools constant, because newer versions of yosys are less efficient, so it's more of a test of whether the synthesized output will still meet timing and fit in the design, which is a nice regression test.

closed 2020-08-15 by xobs