enjoy-digital/litex #641: Memory test fails on Genesys Kintex with BlackParrot

scanakci, 2020-08-24:
Unfortunately, memory tests fail when I use BP with most recent LiteX. I provide the commits when I could successfully pass DRAM tests. I tested several frequencies (75MhZ and 84MhZ do not cause any time violations)

LiteX: c136113a9b71cbcbdf525aaad38acb012f4a12f3
LiteDRAM: fe478382e16ff3592e07774580c96bde0dc82da3
litex-boards: c7404e356f737a58be4527b3ae8de20fce96defd
blackparrot: 7eb1037637d8515a259e204117b7b1273b1c2941

Simulation works perfectly fine (i.e. memory tests pass). At least, being able to reproduce this issue on simulation would be great. 

I shared the BlackParrot repo with @enjoy-digital. In case needed, I can attach here as well.

enjoy-digital, 2020-08-24:
The issue seems to be a calibration issue, but i've not yet been able to reproduce it to investigate (the CPU was not booting). I'll do another test in the next days. The calibration is only specific to the hardware test, so that's not really possible to reproduce this issue on simulation. I'm working on a LiteDRAM bench to make sure we cover all calibration corner cases: https://github.com/enjoy-digital/litedram/tree/master/bench

scanakci, 2020-08-24:
Thanks. I will try to generate bitstream in a fresh vm in the next days and let you know.

developandplay, 2021-06-25:
I think with #953 we can also close this one.
Maybe @enjoy-digital or @scanakci can try if it works on the Genesys too.

enjoy-digital, 2021-06-26:
Thanks @developandplay, I'm also pretty sure your fixes also fix this one. We can close this and I'll do a test on a Genesys2 when I'll have the occasion.

closed 2021-06-26 by enjoy-digital