enjoy-digital/litex #648: Install issue

fallen, 2020-09-08:
Hi, 
I think there is either a setup.py missing or litex_setup.py is out of date because it tries to run setup.py in litex/.
See:
```
[installing litex]...
python3: can't open file 'setup.py': [Errno 2] No such file or directory
Traceback (most recent call last):
  File "./litex_setup.py", line 167, in <module>
    subprocess.check_call("python3 setup.py develop", shell=True)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop' returned non-zero exit status 2.
```

enjoy-digital, 2020-09-09:
Hi @fallen,
i think your issue is similar to https://github.com/enjoy-digital/litex/issues/525#issuecomment-629245574, can you check?

fallen, 2020-09-09:
You are right, I was running from inside the repo.
Maybe the script can be made to work when run from inside the repo?
Running it from outside clones lots of stuff in current directory.
That could be great if the cloning would be circumscribed to only the litex/ top repository.
Thanks!

closed 2020-09-09 by fallen

enjoy-digital, 2020-09-10:
With https://github.com/enjoy-digital/litex/commit/c169494793d63afb789042a913b84477d794b6ef, it's now possible to run it from the cloned repository, but others dependencies will still be cloned alongside.