enjoy-digital/litex #670: OSError: Subprocess failed

bkappa013, 2020-10-08:
Goodmorning` everyone.
I have a problem with litex.
When I try to build the bitstream I get the error listed below.
Without the --build command it does everything smoothly, but I need to have a valid bitstream to load in fpga.
Can someone help me? thank you

`Traceback (most recent call last):
  File "./versa_ecp5.py", line 160, in <module>
    main()
  File "./versa_ecp5.py", line 153, in main
    builder.build(**builder_kargs, run=args.build)
  File "/home/ftecce/litex/litex/soc/integration/builder.py", line 214, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/ftecce/litex/litex/soc/integration/soc.py", line 1048, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/ftecce/litex/litex/build/lattice/platform.py", line 39, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/ftecce/litex/litex/build/lattice/diamond.py", line 238, in build
    _run_script(script)
  File "/home/ftecce/litex/litex/build/lattice/diamond.py", line 151, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed`

enjoy-digital, 2020-10-08:
Hi @bkappa013,

which toolchain are you willing to use for the Versa ECP5: Diamond or Yosys/NextPnr? For your issue, the related toolchain is probably not installed correctly. Can you verify it's in your `PATH`?

mithro, 2020-10-08:
@enjoy-digital We should improve the `raise OSError` at https://github.com/enjoy-digital/litex/blob/004924a3190f5d6b58a213f499e7f8460f9f42c6/litex/build/lattice/diamond.py#L151 to be more informative about what actually failed.

enjoy-digital, 2020-11-04:
@bkappa013, @mithro: thanks for the feedback, this has been improved with https://github.com/enjoy-digital/litex/commit/db836e8e5d9d66caade9294f1a8b67eb2b4d3611.

closed 2020-11-04 by enjoy-digital