enjoy-digital/litex #676: Debugging LiteX / VexRiscv with Verilator / OpenOCD / GDB

lindemer, 2020-10-16:
Is it possible to load a binary onto an emulated LiteX / VexRiscv SoC in Verilator and debug it with GDB? If so, is there a tutorial on how to do this? I'm new to the LiteX ecosystem, and I see that `lxsim` has a `--sim-debug` flag, but I can't find any documentation on it. (I would like to do a cycle-accurate simulation of Zephyr RTOS with a custom VexRiscv plugin.)

Dolu1990, 2020-10-16:
I don't know if there is a TCP to JTAG bridge in litex, but in case of, there is how it can be done in SpinalSim : 
https://github.com/SpinalHDL/SpinalHDL/blob/cd04d9c3b06f82d18c28db9f82f7925276f41745/lib/src/main/scala/spinal/lib/com/jtag/sim/JtagTcp.scala#L10

You just need to open a TCP socket, apply each byte to the JTAG pin directly, and eventualy send back the TDO value as a byte

mithro, 2020-10-16:
@tcal-x You had this question to right?

tcal-x, 2020-10-16:
@mithro I haven't yet had a need to dive into Verilator simulation, but it is useful to know that it exists.   I'd more likely use Verilator for collecting waveforms, but instead use Renode + custom instructions for functional debugging if the new instructions can be handled by the [extension capability](https://github.com/renode/renode/blob/master/tests/unit-tests/riscv-custom-instructions.robot).

tcal-x, 2020-10-16:
> @mithro I haven't yet had a need to dive into Verilator simulation, but it is useful to know that it exists. I'd more likely use Verilator for collecting waveforms, but instead use Renode + custom [...]

...although if I can't find the problem using Renode, then I'd need to get down to this level of detail -- debugging on the actual Verilog implementation using Verilator and the debug bridge with GDB.

lindemer, 2020-12-18:
@enjoy-digital Is there any documentation on this available? The only information I can find is here https://github.com/timvideos/litex-buildenv/wiki/Debugging but I couldn't get this to work on my Arty board. After the `load` command in GDB, the connection appears to break (i.e., I can't read memory or run `continue`).

tcal-x, 2020-12-18:
Hi @lindemer , I replied to you on VexRiscv Gitter, but I have one more comment.   I see the instructions that you pointed to don't use wishbone-tool.    Wishbone-tool is useful for wrapping the openOCD command and making it simpler to use.

From my notes, here is the set of things that I ran, using etherbone:
* set up the laptop's ethernet port to either 192.168.1.100 or 192.168.100.100 (to be the same subnet as the Arty board, whatever that is)  (and replace the `enx9cebe8612341` with your ethernet device name):
```
sudo ip addr add 192.168.1.100/24 dev enx9cebe8612341
sudo ip link set dev enx9cebe8612341 up
```
* Use lxterm/litex\_term from one window with --kernel to load the firmware and interact with it
```
lxterm --speed 115200 --kernel firmware.bin /dev/ttyUSB1
```

* Connect wishbone-tool (I recommend getting the github source and compiling to get the most recent bugfixes).  Again match the address to the IP address built into the Arty board LiteX build.
```
wishbone-tool --ethernet-host 192.168.1.50 -s gdb
```
* Finally start gdb; note that you're using the `.elf`, not the `.bin`:
```
riscv64-unknown-elf-gdb firmware.elf -ex "target remote localhost:3333"
```

I don't (yet) have any experience debugging with a Verilator simulation.

lindemer, 2020-12-21:
@tcal-x This is exactly what I'm looking for - had no idea `wishbone-tool` existed. Won't be able to test this until I'm back from the holidays, but thanks a lot for the tutorial.

tcal-x, 2020-12-21:
I should have added these earlier:
* https://wishbone-utils.readthedocs.io/en/latest/
* https://github.com/litex-hub/wishbone-utils 

Also, there is a section in the Fomu workshop that describes using wishbone-tool's gdb bridge:
* https://workshop.fomu.im/en/latest/riscv.html#debugging-risc-v-code

lindemer, 2021-01-07:
@tcal-x `wishbone-tool` throws this error when I follow your instructions:
```
Error: "invalid configuration: invalid ethernet address: io error invalid socket address"
```
I also tried running `lxsim --cpu-type=vexriscv --cpu-variant=secure+debug --with-etherbone` instead of using the FPGA but I get the same result, even though `lxsim` has automatically generated a new network interface with address `192.168.1.100/24`. Have you encountered this problem?

tcal-x, 2021-01-11:
@lindemer try doing `ip addr` to make sure the network interface is set up correctly.

I don't recall seeing that error.

Are you building your own wishbone-tool?

lindemer, 2021-01-11:
@tcal-x I'm using a USB-Ethernet dongle and as far as I can tell everything's configured as it should be. I installed the latest Rust compiler and built `wishbone-tool` from source.
```
$ ip addr

3: enx00249b284adc: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:24:9b:28:4a:dc brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.100/24 scope global enx00249b284adc
       valid_lft forever preferred_lft forever

$ ip route

default via 10.0.2.2 dev enp0s3 proto dhcp metric 100
10.0.2.0/24 dev enp0s3 proto kernel scope link src 10.0.2.15 metric 100
169.254.0.0/16 dev enp0s3 scope link metric 1000
192.168.1.0/24 dev enx00249b284adc proto kernel scope link src 192.168.1.100
```

lindemer, 2021-01-20:
@kgugala I'm still getting the same error message from `wishbone-tool` described above after pulling your fix for this in https://github.com/enjoy-digital/litex/pull/774. Any ideas?

tcal-x, 2021-01-21:
Hi @lindemer , I just tried running it again, on Arty, using an soc built with `--use-etherbone`, and cpu\_variant="full+debug".  It works, although a bit flakey, but I don't see the error you mention.   So let me dump what I did, and we can see if we missed anything.
```
sudo ip link set dev enx9cebe8612341 down
sudo ip addr add 192.168.1.100/24 dev enx9cebe8612341
sudo ip link set dev enx9cebe8612341 up
```
Note: I noticed my wireless interface was already using 192.168.1.x, so I switched to a different network in my home.

```
$ ip addr
6: enx9cebe8612341: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 9c:eb:e8:61:23:41 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.100/24 scope global enx9cebe8612341
       valid_lft forever preferred_lft forever

$ ip route 
192.168.1.0/24 dev enx9cebe8612341 proto kernel scope link src 192.168.1.100 
```
Wow, almost identical to yours.

Then I run wishbone tool and see the expected response:
```
$ ~/bin/wishbone-tool --ethernet-host 192.168.1.50 -s gdb
INFO [wishbone_tool::server] accepting gdb connections on 127.0.0.1:3333
```

I'm using a top Python script for LiteX that's custom, but I don't know how that would allow it to work for me and not you.   I'll see if I can generate a "pure" LiteX SoC and get it to work, time allowing.

tcal-x, 2021-01-21:
Hmmm, using pure LiteX (with litex\_setup.py), after updating all of the repos, I built with this: `./arty.py --cpu-type=vexriscv --cpu-variant=full+debug --build --with-etherbone --load`.

Now, I cannot connect with wishbone-tool.   I don't get the error message, I just get nothing -- no response.  I need to ctrl-C to exit.

I _can_ ping the Arty board:
```
$ ping 192.168.1.50
PING 192.168.1.50 (192.168.1.50) 56(84) bytes of data.
64 bytes from 192.168.1.50: icmp_seq=1 ttl=128 time=0.899 ms
64 bytes from 192.168.1.50: icmp_seq=2 ttl=128 time=0.490 ms
64 bytes from 192.168.1.50: icmp_seq=3 ttl=128 time=0.425 ms
64 bytes from 192.168.1.50: icmp_seq=4 ttl=128 time=0.418 ms
```
In the other build, where it was working, I had rebuilt the VexRiscv RTL.   I guess it's a possibility that that is the difference.  But you shouldn't have to go down that road.

lindemer, 2021-01-22:
@tcal-x Really appreciate all your help on this. So I've updated LiteX and run the commands you suggested, and I can now at least ping the board at `192.168.1.50`, which is an improvement. I'm still getting the same `invalid socket` error from `wishbone-tool`, though.

I built the tool from source with the latest Rust/Cargo... any chance you're using an older version of it?

tcal-x, 2021-01-22:
@lindemer , I'm using a locally-built wishbone-tool; I'm at this commit:
```
commit 8d1e09b807a7a79ebc1fc2dee8d40a208c788a5b (HEAD -> master, origin/master, origin/HEAD)
Merge: f00e82a b9531c0
Author: Sean Cross <sean@xobs.io>
Date:   Wed Nov 25 10:46:43 2020 +0800

    Merge pull request #34 from betrusted-io/master
    
    change the csr.csv register for the CPU reboot after USB update
```

Hmm, have you checked your firewall settings?

enjoy-digital, 2021-09-08:
Closing since has been answered. This has been labelled as `add-answer-to-wiki` and answer will be added to Wiki.

closed 2021-09-08 by enjoy-digital