enjoy-digital/litex #683: Improve the performance of EtherBone bridge

jedrzejboczar, 2020-10-22:
I was using the EtherBone host bridge to transfer large amount of data between PC and Arty A7 with 100Mbit Ethernet and experienced some performance issues. While write performance is sensible, read performance is very slow and it would be useful to optimize it https://github.com/antmicro/litex-rowhammer-tester/issues/4.

I did some tests using LiteX `RemoteClient`. I was sending/receiving data with incremental addressing in largest possible chunks, i.e. 255 32-bit words per a call to `.write()`/`.read()`. When using a larger number per call, packet encoding fails.

The transfer speeds I could achieve this way were:
* For write of 0x100000 words = 4 MB => ~1.9 MBps
* For read of 0x10000 words = 256 KB => 52.5 KBps 

I did some profiling on the client side using `cProfile`. 
When writing most of the time is spent on encoding the packets in Python code. The actual call to `socket.sendall` takes ~0.24% of the time (at least on the client side). When reading we spend ~80% time in the `socket.recv` call.

I suspect that most of the time is spent in `litex_server` processing the packets. Where would be a good place to start investigating the issue?

jedrzejboczar, 2020-10-22:
By increasing buffer depths of `LiteEthEtherboneRecordSender` and `LiteEthEtherboneRecordReceiver` to 256
(https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/etherbone.py#L285) and changing max_lengh for UDP to 256 https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_server.py#L126 I was able to achieve ~200 KBps read performance. In this case we spend ~43% time in `socket.recv`, so Python encoding/decoding performance starts to play a role here.

enjoy-digital, 2020-10-22:
Thanks for the analysis @jedrzejboczar. Performance has never been optimized on LiteX Server/Etherbone since was good enough for the actual use cases but it would be indeed be good to improve it.

There are probably 3 things to look at first:
1) The encoding/decoding of the Etherbone packets done in Python. Alternative implementations done in C are a lot faster (https://github.com/litex-hub/wishbone-utils/tree/master/libeb-c for example), so it would be good to see if we can improve performance of this encoding/decoding.
 2) The round-trip between commands and responses. This is inherent to the system, but we can reduce the effects by grouping accesses in a same Etherbone packet. The current software/gateware is currently limited to 4 accesses per packet:
- https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/etherbone.py#L202
- https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/etherbone.py#L285
- https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_server.py#L126
This value was chosen to reduce resource usage but could be increased for performance.
3) The current `read_merger`function (https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_server.py#L21) is grouping reads detected as sequential/fixed bursts. This was done for the UARTBone bridge and is also used on Etherbone, but this is a bit restrictive for Etherbone since all pending reads could in fact be regrouped (since the full address is passed for each access in Etherbone), so we could also improve this point. (`read_merger` and `comm_udp.read`).

The writes are probably currently affected by 1) and reads by 1) 2) 3) but 2) 3) are masking the effects of 1).

enjoy-digital, 2020-11-26:
@jedrzejboczar: I've been looking at this yesterday and this morning.

First i created simple benches in LiteEth to simplify testing LiteEth on hardware and do these kind of tests: https://github.com/enjoy-digital/liteeth/tree/master/bench. I also allowed using CommUDP directly from user-scripts (in addition to LiteX Server + Remote Client):

For example:
`./arty.py --build --load`

Then:
`./test_etherbone.py --ident --sram --speed` for testing with LiteX Server  / RemoteClient.
`./test_etherbone.py --ident --sram --speed --udp` for testing directly with CommUDP.

With a `read_merger` configure to `255`, using CommUDP directly already provides interesting speed up on reads:
````
./test_etherbone.py  --speed
Testing write speed...   809.58 KiB/s
Testing read speed...     70.65 KiB/s
````

````
./test_etherbone.py  --speed --udp
Testing write speed...   805.00 KiB/s
Testing read speed...    257.40 KiB/s
````

I also did some simple optimizations in the Etherbone encoding/decoding code that gives 4X speed up:

````
./test_etherbone.py  --speed --udp
Testing write speed...  2431.04 KiB/s
Testing read speed...    913.83 KiB/s
````

Since Etherbone over UDP is mainly used for control/status of the SoC, I think the performance is now reasonable and a good good trade-off between software complexity/performance. For more performance I would recommend switching to C/Rust code for the Etherbone code run on the Host or if you need to move large chunks of data, just do it over UDP (with DMA and LiteEth Streamers modules: https://github.com/enjoy-digital/liteeth/blob/master/liteeth/frontend/stream.py).

enjoy-digital, 2020-12-22:
The main bottlenecks have been addressed and performance improved while keeping simplicity and portability. For more performance, C/Rust code could be used on the Host (Wishbone tool for example) or large blocks of data could be transferred over UDP and DMA.

closed 2020-12-22 by enjoy-digital

ozel, 2024-09-30:
> With a read_merger configure to 255, using CommUDP directly already provides interesting speed up on reads:

So litex_server still does slower reads since the read count is limited to 1 (https://github.com/enjoy-digital/litex/commit/5aa70d975c0d1653d953de36529e2718e940c909), could this be changed now?

For the record, on Butterstick (Gigabit Ethernet, 32-bit data path width) `test_etherbone.py --speed` currently reports up to 10 MiB/s write, and up to 1 MiB/s reads when increasing read count `max_length` in litex_server to 255. Using CommUDP directly (`--udp`), the write speed doesn't increase but reads roughly double to 2 MiB/s. 

The wishbone tools/libraies in `litex-hub/wishbone-utils` also do not support neither write nor read bursts (read/write counts always fixed to 1) in single etherbone packets. But with small changes in litex_server & litex_cli read/write bursts can be supported. Would such a PR be acceptable?

enjoy-digital, 2024-09-30:
@ozel: If configuration of max_length can be made optional from litex_server & litex_cli, this would be a nice contribution yes. (Since some hardware are configured with low buffer_depth to limit resource usage, we cannot use length of 255 by default).