enjoy-digital/litex #686: Wrong repo link for vexriscv_smp

daveshah1, 2020-11-05:
To reproduce, try and build a vexriscv_smp variant SoC without the module installed.
```
ImportError: pythondata-cpu-vexriscv_smp module not installed! Unable to use vexriscv_smp cpu.
No module named 'pythondata_cpu_vexriscv_smp'

You can install this by running;
 pip install git+https://github.com/litex-hub/pythondata-cpu-vexriscv_smp.git
```

But that doesn't work, the link should be https://github.com/litex-hub/pythondata-cpu-vexriscv-smp.git

enjoy-digital, 2020-11-05:
Thanks for the feedback, this is fixed with https://github.com/enjoy-digital/litex/commit/3e47a6e48b3b16db9b24e19143185a3aed0822ce.

closed 2020-11-05 by daveshah1

mithro, 2020-11-05:
@enjoy-digital - I actually think it would have been better to rename the repo to `pythondata-cpu-vexriscv_smp` so you can continue to do things like `_, data_type, data_name = s.split('_')` ?

enjoy-digital, 2020-11-05:
@mithro: indeed thanks, i just did that.