enjoy-digital/litex #687: Building a vexriscv_smp variant SoC standalone?

daveshah1, 2020-11-05:
Running

```
python ~/litex-boards/litex_boards/targets/trellisboard.py --cpu-type vexriscv_smp --cpu-variant linux --build
```

fails with

```
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-11-05 12:06:42)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : LFE5UM5G-85F-8BG756C.
INFO:SoC:System clock: 75.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:uart_phy CSR allocated at Location 3.
INFO:SoCCSRHandler:uart CSR allocated at Location 4.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 5.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:ECP5PLL:Creating ECP5PLL.
INFO:ECP5PLL:Registering Single Ended ClkIn of 12.00MHz.
INFO:ECP5PLL:Creating ClkOut0 sys2x_i of 150.00MHz (+-10000.00ppm).
INFO:ECP5PLL:Creating ClkOut1 init of 25.00MHz (+-10000.00ppm).
INFO:SoCCSRHandler:ddrphy CSR allocated at Location 6.
INFO:SoCCSRHandler:sdram CSR allocated at Location 7.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x40000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoCCSRHandler:leds CSR allocated at Location 8.
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw256
Generating cluster netlist
copying runtime jar...
error: error while loading String, class file '/modules/java.base/java/lang/String.class' is broken
(class java.lang.NullPointerException/null)
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[... 382 lines cut ...]
[error] 	at sbt.MainLoop$.next(MainLoop.scala:139)
[error] 	at sbt.MainLoop$.run(MainLoop.scala:132)
[error] 	at sbt.MainLoop$.$anonfun$runWithNewLog$1(MainLoop.scala:110)
[error] 	at sbt.io.Using.apply(Using.scala:22)
[error] 	at sbt.MainLoop$.runWithNewLog(MainLoop.scala:104)
[error] 	at sbt.MainLoop$.runAndClearLast(MainLoop.scala:59)
[error] 	at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:44)
[error] 	at sbt.MainLoop$.runLogged(MainLoop.scala:35)
[error] 	at sbt.StandardMain$.runManaged(Main.scala:138)
[error] 	at sbt.xMain.run(Main.scala:89)
[error] 	at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:111)
[error] 	at xsbt.boot.Launch$.withContextLoader(Launch.scala:130)
[error] 	at xsbt.boot.Launch$.run(Launch.scala:111)
[error] 	at xsbt.boot.Launch$$anonfun$apply$1.apply(Launch.scala:37)
[error] 	at xsbt.boot.Launch$.launch(Launch.scala:119)
[error] 	at xsbt.boot.Launch$.apply(Launch.scala:20)
[error] 	at xsbt.boot.Boot$.runImpl(Boot.scala:56)
[error] 	at xsbt.boot.Boot$.main(Boot.scala:18)
[error] 	at xsbt.boot.Boot.main(Boot.scala)
[error] java.io.IOError: java.lang.RuntimeException: /packages cannot be represented as URI
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore?
```

Not sure if this is because vexriscv_smp must use linux-on-litex-vexriscv; something is broken in my environment; this is just a vexriscv bug; or litex is calling something wrong?

Dolu1990, 2020-11-05:
Hi,

Basicaly, when the cpu configure asked isn't already pre-generated into the pythondata repo, the flow invoke the VexRiscv generation to fill its netlist cache.

This is done by running the VexRiscv SBT project, and here the issue seem to be related to some issue between SBT and the java environnement installed.

See https://stackoverflow.com/questions/61271015/sbt-fails-with-string-class-is-broken

```
Homebrew maintainers have added a dependency to JDK 13 because they want to use more brew dependencies (brew#50649). This causes sbt to use JDK 13 even when java available on PATH is JDK 8 or 11. To prevent sbt from running on JDK 13, install jEnv or switch to using SDKMAN.
```

What give the following to you ? : 

java --version
javac --version

daveshah1, 2020-11-05:
Looks like I was on OpenJDK 14. Switching to OpenJDK 10 which I also has installed and I get a new failure:

```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/home/david/.sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.7/protobuf-java-3.3.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[info] Loading settings for project all from build.sbt ...
[info] Set current project to VexRiscv (in build file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[info] Updating ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "sim")...
[info] Updating ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "idslpayload")...
[info] Done updating.
[info] Updating ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "idslplugin")...
[info] Done updating.
[info] Done updating.
[info] Updating ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "core")...
[info] Done updating.
[info] Updating ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "lib")...
[info] Done updating.
[info] Updating ...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
fatal: not a git repository: /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/../../../../.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv
[info] Compiling 1 Scala source to /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/idslpayload/target/scala-2.11/classes ...
[info] Compiling 10 Scala sources and 6 Java sources to /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/target/scala-2.11/classes ...
[info] Done compiling.
[info] Compiling 2 Scala sources to /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/idslplugin/target/scala-2.11/classes ...
[info] Packaging /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/idslpayload/target/scala-2.11/spinalhdl-idsl-payload_2.11-1.4.2.jar ...
[info] Done packaging.
[warn] there were 5 deprecation warnings; re-run with -deprecation for details
[warn] one warning found
[info] Done compiling.
[info] Packaging /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/idslplugin/target/scala-2.11/spinalhdl-idsl-plugin_2.11-1.4.2.jar ...
[info] Done packaging.
[warn] there was one deprecation warning; re-run with -deprecation for details
[warn] one warning found
[warn] bootstrap class path not set in conjunction with -source 7
[info] /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java: /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java uses or overrides a deprecated API.
[info] /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java: Recompile with -Xlint:deprecation for details.
[info] /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java: /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java uses unchecked or unsafe operations.
[info] /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/src/main/java/spinal/sim/DynamicCompiler.java: Recompile with -Xlint:unchecked for details.
[info] Done compiling.
[info] Packaging /home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/sim/target/scala-2.11/spinalhdl-sim_2.11-1.4.2.jar ...
[info] Done packaging.
[error] java.lang.RuntimeException: Nonzero exit value: 128
[error] 	at scala.sys.package$.error(package.scala:26)
[error] 	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.slurp(ProcessBuilderImpl.scala:134)
[error] 	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang$bang(ProcessBuilderImpl.scala:104)
[error] 	at $f0a58d4444443cfbe555$.liftedTree1$1(build.sbt:82)
[error] 	at $f0a58d4444443cfbe555$.gitHash(build.sbt:81)
[error] 	at $f0a58d4444443cfbe555$.$anonfun$core$7(build.sbt:143)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:40)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:67)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:269)
[... cut ...]
[error] 	at java.base/java.lang.Thread.run(Thread.java:844)
[error] (ProjectRef(uri("file:/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "core") / Compile / managedSources) Nonzero exit value: 128
```

The `fatal: not a git repository: ` line seems to be the pertinent problem. I installed the pythondata module using

```
pip install git+https://github.com/litex-hub/pythondata-cpu-vexriscv-smp.git
```

Maybe I should have used the `setup.py develop` route instead

Dolu1990, 2020-11-05:
`not a git repository` Hoo right, so, it seem like the clone of https://github.com/litex-hub/pythondata-cpu-vexriscv-smp.git was  wasn't done recursively

Basicaly, in : 
/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/../../../../.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv
And 
/home/david/.local/lib/python3.8/site-packages/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/../../../../.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL

Should be VexRiscv repo and SpinalHDL repo (see https://github.com/litex-hub/pythondata-cpu-vexriscv-smp/tree/master/pythondata_cpu_vexriscv_smp/verilog/ext)

enjoy-digital, 2020-11-05:
pip install is indeed probably not doing the recursive clone. @Dolu1990 can you eventually try with a manual recursive clone of `https://github.com/litex-hub/pythondata-cpu-vexriscv-smp.git` for now and `setup.py develop`? Or with: `./litex_setup.py init `and then `./litex_setup.py install`?

daveshah1, 2020-11-05:
I can confirm that the `setup.py develop` route works as expected

mithro, 2020-11-05:
Currently looks like https://github.com/litex-hub/pythondata-cpu-vexriscv-smp is generated by hand rather than using the https://github.com/litex-hub/pythondata-auto tool to generate it?

Dolu1990, 2020-11-05:
@mithro Yes, i wasn't aware of the flow, so i copied the none smp one and modified by hand.

mithro, 2020-11-05:
@Dolu1990 -- I had a todo item to fix up `pythondata-auto` tool to rebuild the VexRISCV verilog as part of the auto-generation flow.

mithro, 2020-11-05:
@Dolu1990 -- Added https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/issues/1 to track getting pythondata-cpu-vexriscv_smp be autogenerated.

mithro, 2020-11-05:
I created https://github.com/litex-hub/pythondata-auto/issues/4 to track adding auto-generation of verilog as part of the pythondata-auto tool.

mithro, 2020-11-05:
@Dolu1990 - I also created https://github.com/litex-hub/pythondata-auto/issues/5 about the README not saying that the repository is generated using the https://github.com/litex-hub/pythondata-auto tool.

enjoy-digital, 2022-01-07:
We can probably close this since issues have been created in the specific repositories.

closed 2022-01-07 by enjoy-digital