enjoy-digital/litex #693: Finish Microwatt IRQ support

enjoy-digital, 2020-11-09:
Initial IRQ support for Microwatt has been added with https://github.com/enjoy-digital/litex/pull/677 but it seems there seem to be integration issues:

When building with IRQ enabled:
`./arty.py --cpu-type=microwatt --cpu-variant=standard+irq`
raises:
`litex/soc/cores/cpu/microwatt/irq.h:38:22:` error: ‘HOSTXICSICP_BASE’ undeclared (first use in this function)`

and with GHDL:
`./arty.py --cpu-type=microwatt --cpu-variant=standard+ghdl+irq`
raises:
`litex/soc/cores/cpu/microwatt/xics_wrapper.vhdl:41:27:error: no declaration for "ics_to_icp_t"`

madscientist159, 2020-11-09:
Ah, right, I'd forgotten about a small workaround that had to be implemented.  I'm not sure the correct way to handle it, but to get this to work (since the ICS controller has to be exposed as a peripheral on the SoC) the following snippet was needed in versa_ecp5.py:

class BaseSoC(SoCCore):
+    mem_map = {
+        "hostxicsicp": 0xc3ff0000,
+        "hostxicsics": 0xc3ff1000
+    }
+    mem_map.update(SoCCore.mem_map)

madscientist159, 2020-11-11:
@enjoy-digital I suppose the `mem_map` section shown above isn't actually needed, we just tend to use an explicit peripheral memory map in our designs.  I'm not sure how to do the other `add_slave` bits though in the CPU files vs. the SoC target files -- any hints? :smile:

enjoy-digital, 2020-12-30:
@madscientist159: sorry for the delay, the XICS integration has been done with https://github.com/enjoy-digital/litex/commit/7bcebf4cdd34704f6e379e9d5e8da7d7fb8c82bc (this will have to be improved in the future but should be enough for now, this could be useful to have a generic method that will allow the CPUs to add peripherals to the SoC as we are doing here, this would also be useful for VexRiscv-SMP for the PLIC and CLINT).

The issue with missing `ics_to_icp_t` is still present on my side, I'll have a look at this.

enjoy-digital, 2020-12-30:
https://github.com/enjoy-digital/litex/issues/753 has been created to improve the integration.

enjoy-digital, 2020-12-30:
The IRQ version is now working with https://github.com/enjoy-digital/litex/commit/b9e0c95c1836f685386a6f24004882a52c98a120.

This has been tested with:
`/arty.py --cpu-type=microwatt --cpu-variant=standard+irq --integrated-rom-size=0x10000 --build --load`

```

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 30 2020 15:59:16
 BIOS CRC passed (fb76e85d)

 Migen git sha1: d42aa6f
 LiteX git sha1: 74844db

--=============== SoC ==================--
CPU:		Microwatt @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000011111111111111100000| delays: 19+-07
  m0, b2: |00000000000000000000000000001111| delays: 30+-02
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 19+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000011111111111111000000| delays: 19+-07
  m1, b2: |00000000000000000000000000001111| delays: 30+-01
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 19+-06
Switching SDRAM to hardware control.
Memtest at 0x0000000040000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
Memtest OK
Memspeed at 0x0000000040000000 (2MiB)...
  Write speed: 32MiB/s
   Read speed: 54MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex>
```
Thanks @madscientist159.

closed 2020-12-30 by enjoy-digital

mithro, 2020-12-30:
FYI - @shenki / @antonblanchard

Now that we have LiteX + mor1k upstream Linux kernel, maybe we could extend that to also support the PPC architecture?