enjoy-digital/litex #694: Questions about DMA with liteDRAM

andrewb1999, 2020-11-10:
I am experimenting with using the LiteDRAM DMAReader and DMAWriter.  Eventually I want to use them to move data to an HLS generated accelerator, but for now I just want to connect to a sync fifo.  I am having some issues where it seems like the DMA reader is only moving one element into the sync fifo before hanging.  I'm wondering if I could get any input/help here to see if I my gateware and firmware are reasonable.  I am running on a nexys video board using a modified nexys video BaseSoC if that helps.  Here are snippets of my code:

Gateware:   
```
self.submodules.mm2s = mm2s = LiteDRAMDMAReader(self.sdram.crossbar.get_port(), fifo_buffered=True)
mm2s.add_csr()
self.add_csr("mm2s")
self.submodules.s2mm = s2mm = LiteDRAMDMAWriter(self.sdram.crossbar.get_port(), fifo_buffered=True)
s2mm.add_csr()
self.add_csr("s2mm")

self.submodules.sync_fifo = sync_fifo = SyncFIFO([("data", 128)], 100, True)
self.comb += mm2s.source.connect(sync_fifo.sink)
self.comb += sync_fifo.source.connect(s2mm.sink)
```

Firmware:
```
static volatile uint32_t TxBufferPtr[100]  __attribute__((aligned(16)));
static volatile uint32_t RxBufferPtr[100]  __attribute__((aligned(16)));

static void dma_test(void) {

  for(int i = 0; i < 100; i++) {
    TxBufferPtr[i] = i+1;
  }

  for(int i = 0; i < 100; i++) {
    RxBufferPtr[i] = 0;
  }

  flush_l2_cache();

  printf("TxBufferPtr = %#8X\n", (uint32_t)TxBufferPtr);
  printf("RxBufferPtr = %#8X\n", (uint32_t)RxBufferPtr);

  mm2s_base_write((uint32_t)TxBufferPtr);
  mm2s_length_write(25);
  mm2s_start_write(1);
  printf("Waiting for mm2s to finish\n");
  while (!mm2s_done_read());

  busy_wait(1000);

  s2mm_base_write((uint32_t) RxBufferPtr);
  s2mm_length_write(25);
  s2mm_start_write(1);
  printf("Waiting for s2mm to finish\n");
  while(!s2mm_done_read());

  busy_wait(1000);

  int matching = 1;
  for (int i = 0; i < 100; i++) {
    if (RxBufferPtr[i] != TxBufferPtr[i]) {
      printf("Data mismatch at i=%d, expected=%d, actual=%d\n", i, TxBufferPtr[i], RxBufferPtr[i]);
      matching = 0;
    }
  }

  if (!matching) {
    printf("TEST FAILED\n");
  } else {
    printf("TEST PASSED\n");
  }
}
```

Thanks in advance!

enjoy-digital, 2020-11-12:
Hi @andrewb1999,

I'm planing to add examples for this and will try give a detailed answer in the next days.

andrewb1999, 2020-11-20:
@enjoy-digital Not to bother you, but do you at least have a basic example I could take a look at?

enjoy-digital, 2020-11-24:
Hi @andrewb1999,

sorry for the delay. I just had a closer look at your code and it seems you express the length in data words while it has to be express in bytes, so 25 words x 128-bit = 400 bytes. Can you try with that? Otherwise the code seems correct.

Florent

enjoy-digital, 2020-12-22:
I think the question has been answered, closing.

closed 2020-12-22 by enjoy-digital