enjoy-digital/litex #695: Add Ibex as supported CPU

mithro, 2020-11-11:
It would be good to support the Ibex - https://github.com/lowRISC/ibex

mithro, 2020-11-11:
Antmicro will probably look at this...

FYI -- @kgugala / @mgielda / @pgielda

enjoy-digital, 2021-02-15:
Initial support has been added quite easily with https://github.com/enjoy-digital/litex/commit/510bda4c9939f9f025c278e2d04214fe8d6769b6 since interfaces are very similar to CV32E40P:
https://twitter.com/enjoy_digital/status/1361276610972966913

We'll now need to add interrupt support and create a `pythondata-cpu-ibex` package.

mithro, 2021-02-15:
@enjoy-digital - I'll take on creating the `pythondata-cpu-ibex` package

mithro, 2021-02-15:
WIP at https://github.com/litex-hub/pythondata-cpu-ibex

mithro, 2021-02-15:
This should now be published at https://pypi.org/project/pythondata-cpu-ibex/

enjoy-digital, 2021-02-15:
Thanks @mithro, we also need OpenTitan: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/ibex/core.py#L205-L209, is it also possible to include it in the repo?

mithro, 2021-02-15:
I created [`pythondata-misc-opentitan`](https://github.com/litex-hub/pythondata-misc-opentitan) see https://pypi.org/search/?q=pythondata-misc-opentitan

It appears this also includes Ibex at [`pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/`](https://github.com/litex-hub/pythondata-misc-opentitan/tree/master/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl) and things [like the hmac RTL at `pythondata_misc_opentitan/resources/hw/ip/hmac/rtl`](https://github.com/litex-hub/pythondata-misc-opentitan/tree/master/pythondata_misc_opentitan/resources/hw/ip/hmac/rtl)

I called it `misc` because I assume that people like @xobs and @bunnie might want to use the non-CPU RTL parts?

mateusz-holenko, 2021-03-03:
FYI: We created a demo running the LiteX BIOS on LiteX/Ibex platform in Renode: https://github.com/renode/renode/blob/c150b382854ff515464220c945a36823a9589536/scripts/single-node/litex_ibex.resc

enjoy-digital, 2021-03-03:
@mateusz-holenko: Nice, thanks for sharing.

enjoy-digital, 2021-04-29:
Initial Ibex support has been added and pythondata package created. We can close this now and will open specific issues for others points when useful.

closed 2021-04-29 by enjoy-digital