enjoy-digital/litex #712: lxsim error: call to member function 'dump' is ambiguous

nalzok, 2020-11-28:
I'm on macOS 10.14, and using Python 3.8.2 in an Anaconda virtual environment named `FPGA`. I installed litex by:

```
$ cd ~/Developer
$ git clone https://github.com/enjoy-digital/litex.git
$ cd litex
$ conda activate FPGA
$ ./litex_setup.py init install --user
```

Afterwards, I did `lxsim --cpu-type=vexriscv`, and an error occured when I did `lxsim --cpu-type=vexriscv`. Please check the error log below:

```
$ lxsim --cpu-type=vexriscv
INFO:SoC:        __   _ __      _  __
INFO:SoC:       / /  (_) /____ | |/_/
INFO:SoC:      / /__/ / __/ -_)>  <
INFO:SoC:     /____/_/\__/\__/_/|_|
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-11-28 17:51:51)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 1.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:RAM main_ram added Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:uart_phy CSR allocated at Location 3.
INFO:SoCCSRHandler:uart CSR allocated at Location 4.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 5.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Finalized SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False
Bus Regions: (3)
[... 207 lines cut ...]
/usr/bin/clang++  -I.  -MMD -I/opt/local/share/verilator/include -I/opt/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/usr/local/include/ -I/Users/nalzok/Developer/litex/litex/build/sim/core   -c -o Vsim_VexRiscv__1__Slow.o Vsim_VexRiscv__1__Slow.cpp
/usr/bin/clang++  -I.  -MMD -I/opt/local/share/verilator/include -I/opt/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/usr/local/include/ -I/Users/nalzok/Developer/litex/litex/build/sim/core   -c -o Vsim__Dpi.o Vsim__Dpi.cpp
/usr/bin/clang++  -I.  -MMD -I/opt/local/share/verilator/include -I/opt/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/usr/local/include/ -I/Users/nalzok/Developer/litex/litex/build/sim/core   -c -o Vsim__Trace.o Vsim__Trace.cpp
/usr/bin/clang++  -I.  -MMD -I/opt/local/share/verilator/include -I/opt/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/usr/local/include/ -I/Users/nalzok/Developer/litex/litex/build/sim/core   -c -o Vsim__Syms.o Vsim__Syms.cpp
/usr/bin/clang++  -I.  -MMD -I/opt/local/share/verilator/include -I/opt/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/usr/local/include/ -I/Users/nalzok/Developer/litex/litex/build/sim/core   -c -o Vsim__Trace__Slow.o Vsim__Trace__Slow.cpp
/Users/nalzok/Developer/litex/litex/build/sim/core/veril.cpp:75:10: error: call to member function 'dump' is ambiguous
    tfp->dump(main_time);
    ~~~~~^~~~
/opt/local/share/verilator/include/verilated_vcd_c.h:391:10: note: candidate function
    void dump(vluint64_t timeui) { m_sptrace.dump(timeui); }
         ^
/opt/local/share/verilator/include/verilated_vcd_c.h:394:10: note: candidate function
    void dump(double timestamp) { dump(static_cast<vluint64_t>(timestamp)); }
         ^
/opt/local/share/verilator/include/verilated_vcd_c.h:395:10: note: candidate function
    void dump(vluint32_t timestamp) { dump(static_cast<vluint64_t>(timestamp)); }
         ^
/opt/local/share/verilator/include/verilated_vcd_c.h:396:10: note: candidate function
    void dump(int timestamp) { dump(static_cast<vluint64_t>(timestamp)); }
         ^
1 error generated.
make[1]: *** [veril.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [sim] Error 2
```

enjoy-digital, 2020-12-01:
@nalzok: The issue seems specific to the environment or Verilator version (works fine with Ubuntu 18.04 + Verilator 4.013 + GCC). Are you able to test the compilation with GCC instead of Clang?

nalzok, 2020-12-01:
> Are you able to test the compilation with GCC instead of Clang?

I would love to try that out, but how do I specify the toolchain? Using `CC=gcc CXX=g++ lxsim --cpu-type=vexriscv` does not seem to work, i.e. the script still invokes `clang++` and produces the same error as above.

nickoe, 2021-02-25:
I get the exact same problem with `Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b` on a macos image, and I just updated with brew. I guess there is a more fundemental issue somehow? Header files?

nickoe, 2021-02-25:
This appears to be caused by this commit, I guess. https://github.com/enjoy-digital/litex/commit/38054874ac1dc80f75faca9b9af590a4754b9fff

In that commit the argument to dump was an `unsigned int ticks` but it is now using `uint64_t main_time`. So I just tried to cast main_time to an unsigned int and it could build just fine.

```
diff --git a/litex/build/sim/core/veril.cpp b/litex/build/sim/core/veril.cpp
index ee435d37..6357bc0a 100644
--- a/litex/build/sim/core/veril.cpp
+++ b/litex/build/sim/core/veril.cpp
@@ -72,7 +72,7 @@ extern "C" void litex_sim_tracer_dump()
   }
 
   if (dump_enabled && tfp_start <= main_time && main_time <= tfp_end) {
-    tfp->dump(main_time);
+    tfp->dump((unsigned int)main_time);
   }
 }
```

I would certainly not claim I understand this code, but that was just an observation I made. I guess this is not a proper fix, hence I just pasted it here for whoever want to pursue this issue further.

enjoy-digital, 2021-03-03:
Thanks @nickoe for looking at this, casting to `vluint64_t` will probably be more appropriate. Can you try with: https://github.com/enjoy-digital/litex/commit/d3407c67b189d2c338766febb3bf2c57f6f80d1d? @nalzok: Can you also give it a try?

nalzok, 2021-03-03:
I can confirm that the latest commit works now. Kudos to everybody involved!

closed 2021-03-03 by nalzok