enjoy-digital/litex #715: Questionable CV32E40P Performance

rlipperts, 2020-11-30:
In my masters thesis, I'm creating a comparison framework for SoCs. During bare-metal execution of CoreMark on a simple LiteX SoC I noticed that the CV32E40P only reaches ~0.9 Coremark/MHz. In my literature review I found scores that are much higher. On the other hand PicoRV32 scores 0.5 Coremark/MHz and VexRiscv 2.3. These roughly match performance measured elsewhere. All CPUs were built as standard variant.

Is this expected performance (and somehow resulting from the LiteX integration or CPU config used) or a mistake on my side?

The LiteX version i use is up-to-date, CoreMark is freshly cloned from the master branch before execution and i use GCC version 10.1. All CPUs are used in the same SoC config derived from the Basys3 target of the litex-buildtool and run with 100 MHz. Code is loaded to an additional SRAM region specifically included for it. LXterm is used to load the software, vivado for loading the gateware.

enjoy-digital, 2020-11-30:
Thanks for the feedback @rlipperts. I've not used CV32E40P core that much myself so can't answer directly, but I know it's tested in [Embench tester ](https://github.com/antmicro/embench-tester) with test results here: https://antmicro.github.io/embench-tester/. Can you look at Embench tester results and see if this matches your results? (ie  CV32E40P not scoring as expected when integrated in LiteX?) If so we could have a closer look at the LiteX integration.

rlipperts, 2020-11-30:
Sadly, quick research did not yield any Embench scores for CV32E40P in different SoCs. 
I had a look at Embench Testers results and they seem to reflect the same result that I see in Coremark:
CV32E40P has roughly 0.4x the performance of VexRiscv.

Since the CoreMark performance values for CV23E40P I found ([3.19 (but with custom ISA extensions)](https://pulp-platform.org/docs/hipeac/acaces2020/01_Intro_RISC-V.pdf), [3.19, probably just taken from the first source](https://arxiv.org/pdf/2006.14256.pdf)) ist drastically lower something seems to be off. The scores published are not from an independent source (so might be taken with a grain of salt) and do use the XPulp custom extensions, yet the difference is quite big.

enjoy-digital, 2020-11-30:
@rlipperts: would you mind testing with https://github.com/enjoy-digital/litex/commit/d193092e16aad09f48e8849c3bfd4fae915cea0f? I just rewrote the OBI2Wishbone adapter to reduce the write/read latency by 1 clock cycle. This should improve performance.

rlipperts, 2020-11-30:
Indeed it increases the performance to 1.25 Coremark/MHz.

So, for my thesis, the performance discreptancy is most likely a comibnation of a smaller CPU variant used in LiteX (it does not use Xpulp ISA exensions right?) and overhead in the SoC integration?

When I tried to run an automated comparison of performance before and after your commit, both runs got me the same (improved) performance. I noticed that the LiteX setup script always clones the newest versions of the repositories (besides those which are manually specified with an sha1). Wouldn't it be useful if the script always cloned the versions that were used in the corresponding LiteX commit? (And should i create a separate Ticket requesting that?)

enjoy-digital, 2020-11-30:
Indeed, CV32E40P has been integrated in LiteX but it's maybe currently not having it's maximum performance due to some limitation in the integration or LiteX (the fact that Wishbone in LiteX does yet not support bursting could also be another limitation). That's something we could try to improve in the future but I don't think it was the focus of the initial integration work.

Otherwise for the commit versions, we a generally using upstream version of the CPU, unless we have trouble with them which hasn't been the case with CV32E40P for now.

rlipperts, 2020-11-30:
Okay thank you for your help :)

closed 2020-11-30 by rlipperts