enjoy-digital/litex #717: Litex Installation issue on Ubuntu 20.04 with Python 3.8 : Compatible ?

YannQ404, 2020-12-01:
Hello, 

Is the /litex_setup.py init install --xxxx compatible with Ubuntu 20.04 & Python3.8 ? 

I follow the installation guide and have this error :  

[installing migen]...
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'
Traceback (most recent call last):
  File "./litex_setup.py", line 170, in <module>
    subprocess.check_call("python3 setup.py develop", shell=True)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop' returned non-zero exit status 1.

Thanks for your help
Yann

gsomlo, 2020-12-01:
On Tue, Dec 01, 2020 at 05:49:10AM -0800, YannQ404 wrote:
> I follow the installation guide and have this error :
> 
> [installing migen]...
> Traceback (most recent call last):
> File "setup.py", line 4, in
> from setuptools import setup
> ModuleNotFoundError: No module named 'setuptools'

You probably need to have `python3-setuptools` installed before
starting on the installation guide.

YannQ404, 2020-12-01:
That correct the issue. Thanks for the **really** quick reply ;-)

closed 2020-12-01 by YannQ404