enjoy-digital/litex #720: USB-FIFO: understand and fix serialboot issue with sfl_payload_length = 255

enjoy-digital, 2020-12-03:


geertu, 2020-12-04:
Reducing sfl_payload_length is not sufficient for me on OrangeCrab. It still fails with:
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload to device failed due to data corruption (CRC error)

I've bisected the issue, and the first bad commit is commit 513a799a394176ec ("lxterm: Don't attempt to recover from CRC errors during upload.").

enjoy-digital, 2020-12-04:
Thanks @geertu, we you using `--no-crc` before? If so, do you know if it was working without `--no-crc` on the OrangeCrab before (even if slow)?

geertu, 2020-12-04:
I was indeed using --no-crc before, as per the linux-on-litex-vexriscv README.md.
On commit 52a16228951cc69e, I got:

[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (277.4KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (271.3KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (2313 bytes)...
[LXTERM] Upload complete (224.8KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9708 bytes)...
[LXTERM] Upload complete (244.1KB/s).
[LXTERM] Booting the device.

On the last working commit 8eecbd7b575db4bd, without --no-crc, download works, but a little bit slower:

 [LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (189.4KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (190.0KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (2313 bytes)...
[LXTERM] Upload complete (205.6KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9708 bytes)...
[LXTERM] Upload complete (192.1KB/s).

enjoy-digital, 2020-12-04:
@geertu: I've been able to reproduce the issue and fix it with https://github.com/enjoy-digital/litex/commit/894802d13192cc228871087d274f6ed7aed0bb77. Would you mind doing a test? To achieve the maximum speed, we need to set `sfl_payload_length` to `255` but this is not yet possible since not working in all configurations, but 255 seems to be working with the OrangeCrab.

geertu, 2020-12-04:
@enjoy-digital: Thanks, both 894802d13192cc22:

[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (176.1KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (177.2KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (2313 bytes)...
[LXTERM] Upload complete (180.5KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9708 bytes)...
[LXTERM] Upload complete (179.0KB/s).

and b7c0922ec128ed52 are working for me:

[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (176.5KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (177.2KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (2313 bytes)...
[LXTERM] Upload complete (178.0KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9708 bytes)...
[LXTERM] Upload complete (175.1KB/s).

The last commit doesn't seem to make a difference for me.
But in general, it's a bit slower than before.

enjoy-digital, 2020-12-04:
Thanks @geertu for the feedback. For now I first want to make sure the new code is working correctly for all configurations. I'll do some speed tests on the OrangeCrab soon.

enjoy-digital, 2020-12-04:
@geertu: with https://github.com/enjoy-digital/litex/commit/bed072ef19897aae2853cfe4a6c66a6f66701e15 you will get back the previous speed on the OrangeCrab (with an additional CRC check :)).

geertu, 2020-12-04:
@enjoy-digital: Confirmed:

[LXTERM] Uploading buildroot/Image to 0x40000000 (5771252 bytes)...
[LXTERM] Upload complete (211.3KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (204.6KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (2313 bytes)...
[LXTERM] Upload complete (222.0KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9708 bytes)...
[LXTERM] Upload complete (224.9KB/s).

However, I had to change the lxterm port parameter from "/dev/serial/by-id/usb-GsD_OrangeCrab_CDC-if00" to "/dev/ttyACM0", as the code now checks for the "ttyACM" name to special-case CDC-ACM ports.
Perhaps you can readlink() the port when needed, so it works with all of "/dev/ttyACM*", "/dev/serial/by-id/*", and "/dev/serial/by-path/*"? That still won't handle custom udev rules, though.

And what about non-Linux hosts?

enjoy-digital, 2020-12-04:
@geertu: thanks for the feedback. This is just a quick workaround and won't be kept. I just need to investigate the current issue with the USB-FIFO to be able to use the same settings for all kind of ports.

enjoy-digital, 2021-09-30:
The auto-calibration at start-up introduced with https://github.com/enjoy-digital/litex/commit/80cb53fb04060763152cc4fd5f33bfdef8cff90f and https://github.com/enjoy-digital/litex/commit/56614804095ba03a8afec6270281b3d6981b9604https://github.com/enjoy-digital/litex/commit/56614804095ba03a8afec6270281b3d6981b9604 will now automatically configure the litex_term parameters to the best working values. We can probably close this issue.

closed 2021-09-30 by enjoy-digital