enjoy-digital/litex #726: Building target fails on windows

HendrikMennen, 2020-12-10:
Looks like the path is missing slashes?

[image]
```
INFO:SoCBusHandler:Interconnect: ←[1mInterconnectShared←[0m (←[1m2←[0m <-> ←[1m4←[0m).
make: Entering directory `C:/Users/HendrikMennen/Documents/VHDPlus/Litex/litex-boards/litex_boards/targets/build/de10lite/software/libcompiler_rt'
 CC       umodsi3.o
riscv64-unknown-elf-gcc.exe: error: C:UsersHendrikMennenDocumentsVHDPlusLitexpythondata-software-compiler_rtpythondata_software_compiler_rtdata/lib/builtins/umodsi3.c: No such file or directory
riscv64-unknown-elf-gcc.exe: fatal error: no input files
compilation terminated.
make: *** [umodsi3.o] Error 1
make: Leaving directory `C:/Users/HendrikMennen/Documents/VHDPlus/Litex/litex-boards/litex_boards/targets/build/de10lite/software/libcompiler_rt'
Traceback (most recent call last):
  File "de10lite.py", line 128, in <module>
    main()
  File "de10lite.py", line 121, in main
    builder.build(run=args.build)
  File "c:\users\hendrikmennen\documents\vhdplus\litex\litex\litex\soc\integration\builder.py", line 210, in build
    self._generate_rom_software(not self.soc.integrated_rom_initialized)
  File "c:\users\hendrikmennen\documents\vhdplus\litex\litex\litex\soc\integration\builder.py", line 190, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', 'C:\\Users\\HendrikMennen\\Documents\\VHDPlus\\Litex\\litex-boards\\litex_boards\\targets\\build\\de10lite\\software\\libcompiler_rt', '-f', 'c:\\users\\hendrikmennen\\documents\\vhdplus\\litex\\litex\\litex\\soc\\software\\libcompiler_rt\\Makefile']' returned non-zero exit status 2.
```

xobs, 2020-12-10:
This could be a duplicate of https://github.com/enjoy-digital/litex/issues/287

Do you have `sh.exe` in your PATH?

HendrikMennen, 2020-12-10:
yes I have `sh.exe` in my PATH

xobs, 2020-12-10:
What if you set SHELL=cmd.exe?

```powershell
$env:SHELL = "cmd.exe"
```

HendrikMennen, 2020-12-10:
Doesn't seem to change anything for me

enjoy-digital, 2020-12-11:
Can you provide the generated files (create an archive of build/de10lite), this will be useful to understand the issue.

HendrikMennen, 2020-12-11:
gateware folder is empty.

software:
[image]

full archive:
[de10lite.zip](https://github.com/enjoy-digital/litex/files/5681839/de10lite.zip)

enjoy-digital, 2020-12-21:
@HendrikMennen: both #726 and #728 are duplicates and I think @leonbeier is your colleague, let's continue the discussion all together in #728.

closed 2020-12-21 by enjoy-digital