enjoy-digital/litex #728: Problem with building on windows

leonbeier, 2020-12-10:
Somehow I got LiteX working on my own pc, but now with a clean windows pc I went through this installation guide I wrote:
https://github.com/leonbeier/LiteX

And with this I get this error:
[image]

Maybe someone can spot the error in the guide. Should be the same procedure like yours.

enjoy-digital, 2020-12-11:
Can you provide the generated files (create an archive of build/de10lite) with the two different machines? This will be useful to understand the issue. Thanks.

leonbeier, 2020-12-11:
[build.zip](https://github.com/enjoy-digital/litex/files/5679443/build.zip)
This is from the new machine. Only the siftware folders are created.

I made an update on my own pc and now I cannot build there as well

leonbeier, 2020-12-11:
``` sh
cygwin warning:
  MS-DOS style path detected: C:\Users\LeonB\Documents\GitHub\LiteX\litex\litex-boards\litex_boards\targets\build\arty\software\libcompiler_rt
  Preferred POSIX equivalent is: /cygdrive/c/Users/LeonB/Documents/GitHub/LiteX/litex/litex-boards/litex_boards/targets/build/arty/software/libcompiler_rt
  CYGWIN environment variable option "nodosfilewarning" turns off this warning.
  Consult the user's guide for more details about POSIX paths:
    http://cygwin.com/cygwin-ug-net/using.html#using-pathnames
make: Entering directory '/cygdrive/c/Users/LeonB/Documents/GitHub/LiteX/litex/litex-boards/litex_boards/targets/build/arty/software/libcompiler_rt'
c:\users\leonb\documents\github\litex\litex\litex\litex\soc\software\libcompiler_rt\Makefile:22: *** target pattern contains no '%'.  Stop.
make: Leaving directory '/cygdrive/c/Users/LeonB/Documents/GitHub/LiteX/litex/litex-boards/litex_boards/targets/build/arty/software/libcompiler_rt'
Traceback (most recent call last):
  File "arty.py", line 146, in <module>
    main()
  File "arty.py", line 139, in main
    builder.build(**builder_kwargs, run=args.build)
  File "c:\users\leonb\documents\github\litex\litex\litex\litex\soc\integration\builder.py", line 210, in build
    self._generate_rom_software(not self.soc.integrated_rom_initialized)
  File "c:\users\leonb\documents\github\litex\litex\litex\litex\soc\integration\builder.py", line 190, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', 'C:\\Users\\LeonB\\Documents\\GitHub\\LiteX\\litex\\litex-boards\\litex_boards\\targets\\build\\arty\\software\\libcompiler_rt', '-f', 'c:\\users\\leonb\\documents\\github\\litex\\litex\\litex\\litex\\soc\\software\\libcompiler_rt\\Makefile']' returned non-zero exit status 2.
```

This is the error I get now on my other pc. I don't know how I messed that up

enjoy-digital, 2020-12-11:
Thanks, this indeed seems to be related to an update of one the dependencies (python? windows?) I will have a look at your archive.

enjoy-digital, 2020-12-21:
@leonbeier: the issue seems similar to https://github.com/enjoy-digital/litex/issues/287, can you have a look at it and try setting `SHELL=cmd.exe` as described in the installation guide: `Note On Windows, it's possible you'll have to set SHELL environment variable to SHELL=cmd.exe`

leonbeier, 2020-12-21:
I already did this
[image]

enjoy-digital, 2020-12-21:
OK, can you try adding `SHELL = C:/WINDOWS/system32/cmd.exe` to libcompiler_rt's Makefile: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libcompiler_rt/Makefile as suggested in https://github.com/enjoy-digital/litex/issues/287#issuecomment-546184770?

leonbeier, 2020-12-21:
Like this?
[image]
If yes this doesn't help

leonbeier, 2020-12-21:
Do you have a windows subsystem?

enjoy-digital, 2020-12-23:
@leonbeier: sorry I'm no longer using Windows personally but other developers are. If you can try editing the Makefile manually and see what is working on your system this could help understand the root cause.

enjoy-digital, 2021-04-29:
Closing since seems related to the machine and we haven't got feedback.

closed 2021-04-29 by enjoy-digital

SLongofono, 2021-05-17:
Having this issue today on a fresh build, windows machine.

I have set the environment variable SHELL to C:\WINDOWS\system32\cmd.exe, and verified with powershell that it is set properly.

SLongofono, 2021-05-17:
Also, stupid question, but how is make being invoked by the build script?  Could this be related to make missing or otherwise being inaccessible from the invoking environment?