enjoy-digital/litex #733: cpu-type=cv32e40p compile error

UA3MQJ, 2020-12-17:
```
./cyclone_ep4ce10.py --cpu-type=cv32e40p --cpu-variant=standard --integrated-rom-size=0x8000

INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-12-18 01:05:25)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : EP4CE10E22C8.
INFO:SoC:System clock: 50.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:uart_phy CSR allocated at Location 3.
INFO:SoCCSRHandler:uart CSR allocated at Location 4.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 5.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:CycloneIVPLL:Creating CycloneIVPLL, speedgrade -7.
INFO:CycloneIVPLL:Registering Single Ended ClkIn of 50.00MHz.
INFO:CycloneIVPLL:Creating ClkOut0 sys of 50.00MHz (+-10000.00ppm).
INFO:CycloneIVPLL:Creating ClkOut1 sys_ps of 50.00MHz (+-10000.00ppm).
INFO:SoCCSRHandler:sdram CSR allocated at Location 6.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x02000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:CycloneIVPLL:Config:
n          : 1
clk0_freq  : 50.00MHz
clk0_divide: 26
clk0_phase : 0.00°
clk1_freq  : 50.00MHz
clk1_divide: 26
clk1_phase : 90.00°
vco        : 1300.00MHz
m          : 26
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Finalized SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
[... 156 lines cut ...]
python3 -m litex.soc.software.mkmscimg bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/sea/risc-v/litex-boards/litex_boards/targets/build/cyclone_ep4ce10/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 17.94KiB 	(56.06%)
RAM usage: 1.63KiB 	(20.41%)

make: Leaving directory '/home/sea/risc-v/litex-boards/litex_boards/targets/build/cyclone_ep4ce10/software/bios'
Traceback (most recent call last):
  File "./cyclone_ep4ce10.py", line 101, in <module>
    main()
  File "./cyclone_ep4ce10.py", line 94, in main
    builder.build(run=args.build)
  File "/home/sea/risc-v/litex/litex/soc/integration/builder.py", line 217, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/sea/risc-v/litex/litex/soc/integration/soc.py", line 1069, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/sea/risc-v/litex/litex/build/altera/platform.py", line 36, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/sea/risc-v/litex/litex/build/altera/quartus.py", line 210, in build
    _build_qsf(
  File "/home/sea/risc-v/litex/litex/build/altera/quartus.py", line 102, in _build_qsf
    qsf.append(tpl.format(lang=language.upper(), path=filename.replace("\\", "/"), lib=library))
AttributeError: 'NoneType' object has no attribute 'upper'

```

Disasm, 2020-12-18:
Additional info: this happens because cv32e40p provides a file with `.svh` extension which is added to the source list with language `None`. For this file LiteX also generates a `load_None` command for yosys. This CPU also fails to build with yosys, because LiteX generates a `load_systemverilog` command for other files that doesn't exist (`load_verilog -sv` should be used instead).

mithro, 2020-12-18:
@kgugala - As @antmicro added this core, can you assign someone to look into this?

kgugala, 2020-12-18:
This seems to be a generic issue with SV sources in Intel targets. I'll get somebody to check it

kgugala, 2020-12-20:
@UA3MQJ where can I find the `cyclone_ep4ce10.py` target? I don't see it neither in LiteX nor in litex-boards

Disasm, 2020-12-20:
@kgugala you can use `./de10lite.py --cpu-type=cv32e40p` to reproduce the issue.

kgugala, 2020-12-20:
I opened #738 fixing the crash reported here. With this one LiteX seems to generate correct qsf. However, Quartus does not seem to be able to elaborate cv32e40p sources - it complains about bunch of syntax errors. I suppose this should be discussed upstream (in cv32e40 repo)

kgugala, 2020-12-20:
I tested this with Quartus 20.1

enjoy-digital, 2020-12-21:
Thanks @kgugala, @Disasm. I agree with @kgugala that we should first be sure the CPU can be compiled correctly with Quartus.

closed 2020-12-21 by enjoy-digital

UA3MQJ, 2020-12-21:
Thanks!
There are no errors by Litex now.
Remain only synthesis errors in Quartus.
```
Error (10170): Verilog HDL syntax error at riscv_id_stage.sv(937) near text: "genvar";  expecting an identifier ("genvar" is a reserved keyword ). Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/sea/risc-v/pythondata-cpu-cv32e40p/pythondata_cpu_cv32e40p/system_verilog/rtl/riscv_id_stage.sv Line: 937
Error (10112): Ignored design unit "riscv_id_stage" at riscv_id_stage.sv(40) due to previous errors File: /home/sea/risc-v/pythondata-cpu-cv32e40p/pythondata_cpu_cv32e40p/system_verilog/rtl/riscv_id_stage.sv Line: 40
Info (12021): Found 0 design units, including 0 entities, in source file /home/sea/risc-v/pythondata-cpu-c
```

PR #738 (2020-12-21): Quartus: handle vh and svh files
fixes #733