enjoy-digital/litex #739: vexriscv_smp: add option for 64 memory width using litedram

devobliquezer0, 2020-12-21:
This issue concerns 3 repositories:
litex, linux-on-litex-vexriscv and pythondata-cpu-vexriscv_smp

Since the introduction of vexriscv_smp in linux-on-litex-vexriscv I have tried building 4 core variant for the nexys4ddr target but the build fails with:
```
ERROR: [Common 17-69] Command failed: File '/build/FPGA/RISCV/litex/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc3_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw128_Cdma.v' does not exist
```
This file is generated by pythondata-cpu-vexriscv_smp based on contents of soc/cores/cpu/vexriscv_smp/core.py under this repo.
Which has Ldw hardcoded to 128 here: https://github.com/enjoy-digital/litex/blob/90b9f4eca3b47e665bf7cbd78df47a9662d332c0/litex/soc/cores/cpu/vexriscv_smp/core.py#L164

Changing this value to 64 and regenerating pythondata solves the issue.

I see two possible solutions, one is to have data_width as an arg, the other is to nest the multicore loop in another loop for all supported data_width values. In the latter case why not add 6 and 8 core count as well along with 2,4? I have tested 8 core on my nexys4ddr and it works fine.

enjoy-digital, 2020-12-21:
Hi @devobliquezer0,

in fact the most common configuration are pre-generated, but when not found LiteX should generate the configuration automatically from SpinalHDL sources, ex you should see:

``
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc4_Iw64Is8192Iy2_Dw64Ds8192Dy2_Ldw64_Cdma
Generating cluster netlist
``

Can you first verify SpinalHDL is installed correctly? Installing LiteX with `litex_setup.py` following the installation guide should be enough for that: https://github.com/enjoy-digital/litex/wiki/Installation

devobliquezer0, 2020-12-21:
> Hi @devobliquezer0,
>
> in fact the most common configuration are pre-generated, but when not found LiteX should generate the configuration automatically from SpinalHDL sources, ex you should see: [...]

Installation looks fine and updated. This error only occurs when I follow design flow from this repo https://github.com/litex-hub/linux-on-litex-vexriscv and just use the command `./make.py --board=nexys4ddr --cpu-count=4 --build` .

If I generate the regular way using:
`litex-boards/litex_boards/targets/nexys4ddr.py --build --cpu-type vexriscv_smp --cpu-variant=4c --cpu-variant linux --with-ethernet --with-sdcard --integrated-rom-size 0x10000`
then it starts generating the correct config using sbt

enjoy-digital, 2020-12-21:
Thanks @devobliquezer0, I will try this and investigate. This could be related to the recent merge of the AES instruction support.

enjoy-digital, 2020-12-21:
@devobliquezer0: Some changes were missing in https://github.com/litex-hub/pythondata-cpu-vexriscv_smp. This has been fixed. Can you go to your local `pythondata-cpu-vexriscv_smp` and do:
```
git pull
git submodule update
```
And then retry to build the nexys4ddr target with the command line you were using?

Dolu1990, 2020-12-21:
@devobliquezer0 If not fixed can you show the complette logs from ? Maybe there is issue with your environnement scala toolchain setup.

devobliquezer0, 2020-12-21:
@enjoy-digital @Dolu1990 yup that fixed it, thanks!

enjoy-digital, 2020-12-21:
Good, thanks for the feedback.

closed 2020-12-21 by enjoy-digital