enjoy-digital/litex #740: nexys4ddr: sdcard logic broken.

devobliquezer0, 2020-12-21:
With all the sdcards I have, few EVOs, SanDisks etc. The sdcard doesn't seem to work at all. It doesn't load images or gets initialized by linux.
I get the following error during devicetree generation:
```
build/nexys4ddr/nexys4ddr.dts:195.22-203.5: Warning (simple_bus_reg): /soc/mmc@f0006000: simple-bus unit address format error, expected "f0005800"
```

During linux boot on vexriscv it gets stuck at:
```
[    2.611768] litex-mmc f0005800.mmc: invalid resource
```

During rocketchip linux boot it loops on this error:
```
[   31.170278] litex-mmc 12005800.mmc: Requested clk_freq=0: set to 195312 via div=256
[   32.279559] litex-mmc 12005800.mmc: Requested clk_freq=12500000: set to 12500000 via div=4
[   32.334506] Command (cmd 8) failed, status 2
[   32.341200] Command (cmd 55) failed, status 2
[   32.347914] Command (cmd 55) failed, status 2
[   32.354816] Command (cmd 55) failed, status 2
[   32.361528] Command (cmd 55) failed, status 2
[   32.368258] Command (cmd 1) failed, status 2
[   32.370177] litex-mmc 12005800.mmc: Requested clk_freq=0: set to 195312 via div=256
[   33.479560] litex-mmc 12005800.mmc: Requested clk_freq=12500000: set to 12500000 via div=4
[   33.534497] Command (cmd 8) failed, status 2
[   33.541194] Command (cmd 55) failed, status 2
[   33.547907] Command (cmd 55) failed, status 2
[   33.554839] Command (cmd 55) failed, status 2
[   33.561555] Command (cmd 55) failed, status 2
[   33.568272] Command (cmd 1) failed, status 2
[   33.570221] litex-mmc 12005800.mmc: Requested clk_freq=0: set to 195312 via div=256
[   34.679560] litex-mmc 12005800.mmc: Requested clk_freq=12500000: set to 12500000 via div=4

```

The only solution to get sdcard working is to use the spi-sdcard logic.

enjoy-digital, 2020-12-21:
Thanks for the feedback @devobliquezer0, can you provide the command you are using to generate the Nexys4DDR bit-stream? Can you also provide the bitstream so that I test locally on my Nexys4DDR?  For the Linux-driver, the .dts generation is probably not correct currently and I was going to look at this.

devobliquezer0, 2020-12-21:
I just used make.py from https://github.com/litex-hub/linux-on-litex-vexriscv

mateusz-holenko, 2020-12-21:
I think that `[    2.611768] litex-mmc f0005800.mmc: invalid resource` might be fixed with https://github.com/antmicro/linux-on-litex-vexriscv/commit/fd0cc0ddfe022d53c38725ab86460e21d541c0f0.

enjoy-digital, 2020-12-21:
Thanks @mateusz-holenko, I will try this and integrate if working (that's one of the last thing I wanted to get working with the VexRiscv-SMP merge but it seems users are faster than me :))

@devobliquezer0: linux-on-litex-vexriscv and linux-on-litex-rocket are two different projects, so it's not possible to mix things between them (unless you are already very familiar with LiteX). So to be able to help, having the exact command you are using + instructions + bitstream if possible would be very useful.

devobliquezer0, 2020-12-21:
> Thanks @mateusz-holenko, I will try this and integrate if working (that's one of the last thing I wanted to get working with the VexRiscv-SMP merge but it seems users are faster than me :)) [...]

ah sorry, i wasn't clear. I did try them both separately.

for linux-vexriscv
```
./make.py --board=nexys4ddr --cpu-count=1 --build
```

for linux-rocket
```
litex-boards/litex_boards/targets/nexys4ddr.py --build --load \
   --cpu-type rocket --cpu-variant linux --sys-clk-freq 50e6 \
   --with-ethernet --with-sdcard --integrated-rom-size 0x10000
```

As i said, switching both to use spi-sdcard works fine.

enjoy-digital, 2020-12-21:
@devobliquezer0: I created https://github.com/litex-hub/linux-on-litex-vexriscv/issues/171 to track progress in SD-Mode and reverted all boards in linux-on-litex-vexriscv to SPI-Mode for now.

closed 2020-12-21 by enjoy-digital