enjoy-digital/litex #743: TypeError: cannot pickle 'weakref' object

hinzundcode, 2020-12-23:
```
$ lxterm --kernel firmware.bin /dev/cu.usbserial-14101 
Traceback (most recent call last):
  File "/Users/chris/Projekte/ecpix5/litex/litex/litex/tools/litex_term.py", line 498, in main
    term.start()
  File "/Users/chris/Projekte/ecpix5/litex/litex/litex/tools/litex_term.py", line 462, in start
    self.start_writer()
  File "/Users/chris/Projekte/ecpix5/litex/litex/litex/tools/litex_term.py", line 454, in start_writer
    self.writer_thread.start()
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: cannot pickle 'weakref' object

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/chris/Projekte/ecpix5/env/bin/lxterm", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'lxterm')())
  File "/Users/chris/Projekte/ecpix5/litex/litex/litex/tools/litex_term.py", line 504, in main
    term.stop()
  File "/Users/chris/Projekte/ecpix5/litex/litex/litex/tools/litex_term.py", line 466, in stop
    self.writer_thread.terminate()
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/process.py", line 133, in terminate
    self._popen.terminate()
AttributeError: 'NoneType' object has no attribute 'terminate'
```

I'm on macOS 11.0.1, litex is on d90d3e0

cr1901, 2021-01-07:
Equivalent error on Windows (_I have some local changes as well that are irrelevant to this issue_):

```
William@DESKTOP-H0PMN4M MINGW64 ~/Projects/FPGA/litex/litex
$ winpty litex_term --speed 38400 COM3
Traceback (most recent call last):
  File "C:/msys64/home/william/projects/fpga/litex/litex/litex/tools/litex_term.py", line 501, in main
    term.start()
  File "C:/msys64/home/william/projects/fpga/litex/litex/litex/tools/litex_term.py", line 464, in start
    self.start_reader()
  File "C:/msys64/home/william/projects/fpga/litex/litex/litex/tools/litex_term.py", line 433, in start_reader
    self.reader_thread.start()
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/context.py", line 326, in _Popen
    return Popen(process_obj)
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/popen_spawn_win32.py", line 93, in __init__
    reduction.dump(process_obj, to_child)
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
ValueError: ctypes objects containing pointers cannot be pickled

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\litex_term-script.py", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'litex_term')()
  File "C:/msys64/home/william/projects/fpga/litex/litex/litex/tools/litex_term.py", line 507, in main
    term.stop()
  File "C:/msys64/home/william/projects/fpga/litex/litex/litex/tools/litex_term.py", line 468, in stop
    self.reader_thread.terminate()
  File "C:/msys64/mingw64/lib/python3.8/multiprocessing/process.py", line 133, in terminate
    self._popen.terminate()
AttributeError: 'NoneType' object has no attribute 'terminate'
```

Seems `multiprocessing` does some weird things on OS X and Windows that makes `lxterm` not work at all. Is it possible to [make](https://stackoverflow.com/questions/8804830/python-multiprocessing-picklingerror-cant-pickle-type-function) `reader` pickle-able?

zyp, 2021-01-11:
Looks like feeb2f7 arbitrarily changed from threading to multiprocessing, and I don't really see what the advantage of using multiprocessing for this would be. The disadvantage is now obvious. :)

I imagine the difference in behavior between the OSes is a result of differences in how child processes are started and how data is shared with them.

enjoy-digital, 2021-01-11:
@cr1901, @hinzundcode: Use of threading has been reverted with https://github.com/enjoy-digital/litex/commit/460fada3acbc6626f15388e4259bec0058b4a2d6 and should solves the issues with Windows and OS X. Sorry for this I was only testing on Linux.

enjoy-digital, 2021-01-13:
I think we can close this since resolved by reverting the code to use threading.

closed 2021-01-13 by enjoy-digital