enjoy-digital/litex #744: litex_json2dts.py doesn't configure interrupts

stffrdhrn, 2020-12-25:
Examples generated by the dts generator

```
            liteuart0: serial@e0002000 {
                device_type = "serial";
                compatible = "litex,liteuart";
                reg = <0xe0002000 0x100>;
                status = "okay";
            };

            mac0: mac@e0004800 {
                compatible = "litex,liteeth";
                reg = <0xe0004800 0x7c>,
                      <0xe0004000 0x0a>,
                      <0x80000000 0x2000>;
                tx-fifo-depth = <2>;
                rx-fifo-depth = <2>;
            };
```

By had I have configured;

```
        serial0: serial@e0002000 {
                device_type = "serial";
                compatible = "litex,liteuart";
                reg = <0xe0002000 0x100>;
                interrupts = <0>;
        };

        mac0: mac@e0006000 {
                compatible = "litex,liteeth";
                reg = <0xe0004800 0x7c
                       0xe0004000 0x0a
                       0x80000000 0x2000>;
                tx-fifo-depth = <2>;
                rx-fifo-depth = <2>;
                interrupts = <2>;
        };
```

The generated DTS blocks the the above devices are missing the interrupts mapping, I think in these cases the drivers will work but fall back to polling for IO.

mithro, 2020-12-26:
FYI - @mateusz-holenko

enjoy-digital, 2020-12-29:
https://github.com/enjoy-digital/litex/pull/745 has been merged and optional polling mode support has been added with https://github.com/enjoy-digital/litex/commit/bf32d23d9a5f6e88a5e7e226b54e0414cf6f1eec, thanks @stffrdhrn.

closed 2020-12-29 by enjoy-digital