enjoy-digital/litex #746: about add hw modules and peripherals

SanadaShinken, 2020-12-26:
Hi, All:

after build a basic SoC configuration, there are two works has to be expands.

1. add our designed verilog based modules into current SoC configuration, and write linux drivers. what I know is using add_ip to add hw design module. but how do I assign resources, like interrupt signal to  interrupt controller, memory address, signal pins, register, etc. is there a sample project could be a good starting point.

2. add peripherals, like three uart ports, two I2C, etc. is there any examples ?

3. add a litex based design into current configuration. I could use fpga_101 as a start point. but I don't know how to assign resource as problem #1.

BR, Sanada

enjoy-digital, 2020-12-28:
Hi @SanadaShinken,

This is a broad question and it's difficult to give a precise answer here. We are planning to add more Wiki pages and documentation for this in the future. For now I'd recommend exploring the actual LiteX's Wiki and projects created with it:
- https://github.com/enjoy-digital/litex/wiki/Tutorials-Resources
- https://github.com/enjoy-digital/litex/wiki/Projects
- https://github.com/enjoy-digital/litex/wiki/LiteX-for-Hardware-Engineers

Since you want to integrate peripherals in a Linux SoC, I would also recommend exploring this part of LiteX:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1074-L1587 that provides convenient methods to add peripherals (UART, SPI, Ethernet, SATA, SDCard, etc...) to a LiteX SoC. 

Once integrated, you'll also have to create Linux drivers for them (you can explore https://github.com/litex-hub/linux/commits/litex-rebase for the current supported peripherals) and  declare the peripherals in your `.dts` file. For this you can explore https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py that we are using to generate custom `.dts` for the [Linux-on-LiteX-Vexriscv](https://github.com/litex-hub/linux-on-litex-vexriscv) project.

Betrusted-IO project can also be interesting to look at to find integration examples of external cores with LiteX: https://github.com/betrusted-io/gateware

SanadaShinken, 2020-12-28:
> Hi @SanadaShinken,
>
> This is a broad question and it's difficult to give a precise answer here. We are planning to add more Wiki pages and documentation for this in the future. For now I'd recommend [...]

Hi @enjoy-digital :

Thank you very much!!!
I'll do a lot of study from you provide. for the current SoC need, UART x3, SPI x 2, video processing module, camera input, usb, hdmi output is very important to our internal plan.

Many Thanks!!

BR, Sanda

enjoy-digital, 2021-06-17:
A reference to this issue has been added to https://github.com/enjoy-digital/litex/wiki/Add-A-New-Core, part of this contents could be reuse to write this wiki page.

closed 2021-06-17 by enjoy-digital