enjoy-digital/litex #748: litex_json2dts.py: Has hard coded initrd settings that are too small

stffrdhrn, 2020-12-28:
The settings for initrd are hard coded in the json file as below.  It would be nice if we could pass sizes in via the command like. For example:  `--initrd-start 0x00800000 --initrd-size 1234567`, the values hard coded now in the script can be used as fall back defaults.

https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py#L34-L42

stffrdhrn, 2020-12-28:
@mateusz-holenko

mateusz-holenko, 2020-12-28:
@stffrdhrn Sounds like a good idea to me, added some comments to #749.

enjoy-digital, 2020-12-28:
Thanks @stffrdhrn, @mateusz-holenko for looking at this. The dicts were just added to avoid breaking mork1x support  when merging VexRiscv SMP, being able to configure this while keeping current default values also sounds like a good idea to me.

enjoy-digital, 2020-12-29:
https://github.com/enjoy-digital/litex/pull/749 has been merged, thanks @stffrdhrn, @mateusz-holenko.

closed 2020-12-29 by enjoy-digital