enjoy-digital/litex #751: Memory initialization failed on Arty A7

dayjaby, 2020-12-29:
Hey,

After running arty.py like this:
```
python3 ./arty.py --toolchain symbiflow --cpu-type vexriscv --sys-clk-freq 80e6 --with-ethernet --load
screen /dev/ttyUSB1 115200 8N1
```

I got address and data errors:

```
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000001| delays: 31+-00
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 31+-00
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000111| delays: 30+-01
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 30+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  0/256
  addr errors: 1372/8192
  data errors: 65614/524288
Memtest KO
Memory initialization failed
```

The only change I did was to set integrated_rom_size to 65536, because the rom was too small for the --with-ethernet option:

```
class BaseSoC(SoCCore):
    def __init__(self, toolchain="vivado", sys_clk_freq=int(100e6), with_ethernet=False, with_etherbone=False, ident_version=True, **kwargs):
        platform = arty.Platform(toolchain=toolchain)
        kwargs["integrated_rom_size"] = 65536

        # SoCCore ----------------------------------------------------------------------------------
        SoCCore.__init__(self, platform, sys_clk_freq,
            ident          = "LiteX SoC on Arty A7",
            ident_version  = ident_version,
            **kwargs)
```

LiteX version: https://github.com/enjoy-digital/litex/commit/bddb170650242e3f993b882806474ca3480a6a7c
Symbiflow version: https://github.com/SymbiFlow/symbiflow-examples/commit/c092a1327423d46a0b31c3389ec34a9240c3f97c

Any hint where to start looking?

mithro, 2020-12-29:
@dayjaby - Did the result meet timing?

enjoy-digital, 2020-12-30:
Hi @dayjaby,

are you able to build the design at 100MHz? If so, I would recommend using this since regularly tested with Vivado. A test you can do is using `ddr3_mr_gen` tool from LiteDRAM: https://github.com/enjoy-digital/litedram/blob/master/bench/ddr3_mr_gen.py and try to change CL to 7:

```
./ddr3_mr_gen.py --cl=7 --cwl=5
DDR3 Timing Settings:
cl:  7
cwl: 5
DDR3 Electrical Settings:
rtt_nom: 60ohm
rtt_wr:  60ohm
ron:     34ohm
Commands to be used with LiteX BIOS:
sdram_mr_write 0 2608
sdram_mr_write 1 2054
sdram_mr_write 2 512
```
and then copy/past the provided command in the BIOS and then run `sdram_cal` and `sdram_test`:

````
       __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 30 2020 09:55:57
 BIOS CRC passed (4ec4c00a)

 Migen git sha1: d42aa6f
 LiteX git sha1: 3e115a0e

--=============== SoC ==================--
CPU:		VexRiscv @ 80MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 640MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000011111111| delays: 28+-04
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 28+-04
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000001111111| delays: 28+-03
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 28+-03
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 21MiB/s
   Read speed: 17MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Cancelled

--============= Console ================--

litex> 
litex> sdram_mr_write 0 2608
Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2054
Switching SDRAM to software control.
Writing 0x0806 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 512
Switching SDRAM to software control.
Writing 0x0200 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |11100000000000000000000000000000| delays: 01+-01
  m0, b2: |00001111111111111111110000000000| delays: 13+-09
  m0, b3: |00000000000000000000000011111111| delays: 28+-04
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 13+-09
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |11100000000000000000000000000000| delays: 01+-01
  m1, b2: |00001111111111111111100000000000| delays: 13+-08
  m1, b3: |00000000000000000000000001111111| delays: 28+-03
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 13+-09
Switching SDRAM to hardware control.

litex> sdram_test
Memtest at 0x40000000 (8MiB)...
  Write: 0x40000000-0x40800000 8MiB     
   Read: 0x40000000-0x40800000 8MiB     
Memtest OK
````

If working we could see if it's interesting to provide a way to force CL/CWL values.

dayjaby, 2020-12-30:
I tried it today again and it suddenly works. I will investigate whether it's some kind of temperature issue as I tested the device yesterday for some hours...

Here the working log:
```
��      __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 29 2020 21:18:11
 BIOS CRC passed (253c1426)

 Migen git sha1: d42aa6f
 LiteX git sha1: bddb1706

--=============== SoC ==================--
CPU:		VexRiscv @ 80MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 640MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000111111| delays: 29+-03
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 29+-03
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000111111| delays: 29+-03
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 29+-03
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 21MiB/s
   Read speed: 17MiB/s
```

bunnie, 2020-12-30:
On NeTV2 I definitely had problems with stability at the cold corner, and the problem was specific to a particular lot of chips. Went through a lot of freeze spray that week...

dayjaby, 2020-12-30:
After rebuilding everything with 100Mhz:
```
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed
```

Setting the SDRAM values helps only slightly:
```
--============= Console ================--

litex> sdram_mr_write 0 2608
Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2054
Switching SDRAM to software control.
Writing 0x0806 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 512
Switching SDRAM to software control.
Writing 0x0200 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000011110000000000000000000| delays: 11+-02
  m0, b3: |00000000000000000000000011100000| delays: 26+-02
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 11+-02
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000011111100000000000000000000| delays: 09+-02
  m1, b3: |00000000000000000000000111110000| delays: 25+-02
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 09+-03
Switching SDRAM to hardware control.

litex> sdram_test
Memtest at 0x40000000 (8MiB)...
  Write: 0x40000000-0x40800000 8MiB     
   Read: 0x40000000-0x40800000 8MiB     
  bus errors:  0/256
  addr errors: 997/8192
  data errors: 367962/2097152
Memtest KO
```

Tested again with 80Mhz and it worked again.

enjoy-digital, 2020-12-30:
Thanks for the feedback @dayjaby, @bunnie. The validity window during the read leveling seems very short compared to the results we generally have with Vivado:

````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 30 2020 14:06:59
 BIOS CRC passed (8d318cbb)

 Migen git sha1: d42aa6f
 LiteX git sha1: 7bcebf4c

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000011111111111111100000| delays: 19+-07
  m0, b2: |00000000000000000000000000001111| delays: 30+-02
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 19+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000011111111111111000000| delays: 19+-06
  m1, b2: |00000000000000000000000000000111| delays: 30+-01
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 19+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 27MiB/s
   Read speed: 22MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> 

````

It could be related to some IOs settings that are not configured correctly with Symbiflow or timings issues and would probably have to be investigated by the Symbiflow team. @dayjaby can you test with the attached bitstream? (same than yours but generated with Vivado): [arty_2020_12_30_vivado_2018.2.zip](https://github.com/enjoy-digital/litex/files/5754315/arty_2020_12_30_vivado_2018.2.zip)

dayjaby, 2020-12-30:
@enjoy-digital looking good with your bitstream:

```
--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000111111111111110000| delays: 21+-07
  m0, b2: |00000000000000000000000000000011| delays: 31+-01
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 21+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000111111111111110000| delays: 21+-07
  m1, b2: |00000000000000000000000000000011| delays: 31+-01
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 21+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 27MiB/s
   Read speed: 22MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x40000000... 
Network boot failed.
No boot medium found

--============= Console ================--

litex> sdram_mr_write 0 2608
Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2054
Switching SDRAM to software control.
Writing 0x0806 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 512
Switching SDRAM to software control.
Writing 0x0200 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |11111111111100000000000000000000| delays: 06+-06
  m0, b3: |00000000000001111111111111100000| delays: 20+-07
  m0, b4: |00000000000000000000000000000111| delays: 30+-01
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 20+-07
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |11111111111100000000000000000000| delays: 06+-06
  m1, b3: |00000000000000111111111111110000| delays: 21+-07
  m1, b4: |00000000000000000000000000000011| delays: 31+-01
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 21+-07
Switching SDRAM to hardware control.

litex> sdram_test
Memtest at 0x40000000 (8MiB)...
  Write: 0x40000000-0x40800000 8MiB     
   Read: 0x40000000-0x40800000 8MiB     
Memtest OK
```

mithro, 2020-12-30:
@dayjaby - I would most likely blame the SymbiFlow toolchain at the moment.

@acomodi / @kgugala - Could you help @dayjaby figure out what is going on here?

andrewb1999, 2021-01-05:
I am also having the same issue as @dayjaby

It seems clear it is an issue with symbiflow as it works with Vivado.

kgugala, 2021-01-06:
@andrewb1999 @dayjaby can you share your build logs?

dayjaby, 2021-02-02:
@kgugala I will try to reproduce the same error now.

Here some other failed build log (https://pastebin.com/Y7B0BCYe) whilst running this command:

```
~/Development/litex/litex-boards/litex_boards/targets$ python3 ./arty.py --toolchain symbiflow --cpu-type vexriscv --cpu-variant=imac+debug --sys-clk-freq 80e6 --with-etherbone --local-ip=192.168.178.244 --remote-ip=192.168.178.25 --csr-csv csr.csv --build > build_output.txt 2>&1
```

With Vivado this one worked.

enjoy-digital, 2021-09-08:
From the investigation that have been done, the issue should probably be tracked in prjxray project. I created https://github.com/SymbiFlow/prjxray/issues/1740 for it.

closed 2021-09-08 by enjoy-digital