enjoy-digital/litex #752: Build bare metal demo app fails

smosanu, 2020-12-29:
I updated LiteX and RISCV tools and managed to run Linux on LiteX VexRiscV on Arty_A7 with 4 cores, which was quite impressive, thank you!
Now, to run bare metal apps, I tried compiling the demo but am stuck at this error:
`linux-on-litex-vexriscv$ litex_bare_metal_demo --build-path=build/arty_a7/
 CC       main.o
 LD       demo.elf
riscv64-unknown-elf-ld: ../build/arty_a7//software/libbase/crt0.o: in function `.L0 ':
/localtmp/EnDigital/litex/litex/soc/cores/cpu/vexriscv_smp/crt0.S:81: undefined reference to `smp_slave'
Makefile:18: recipe for target 'demo.elf' failed
make: *** [demo.elf] Error 1
cp: cannot stat 'demo/demo.bin': No such file or directory`

Looking into the files litex/litex/soc/cores/cpu/vexriscv_smp/crt0.S and litex/litex/soc/cores/cpu/vexriscv_smp/boot-helper.S the `smp_slave` is declared global though only in `boot-helpter.S` so am suspecting a linking error between the two files.

I'll keep digging into it and hope to figure it out...

enjoy-digital, 2020-12-30:
Thanks @smosanu for the feedback. This has been fixed with https://github.com/enjoy-digital/litex/commit/0cba91022e4304684ed37e16d8b3e08912ae4415 and tested on hardware.

smosanu, 2021-01-04:
Thank you! Tested successfully on my end as well. Closing the issue.

closed 2021-01-04 by smosanu