enjoy-digital/litex #753: Allow CPUs to add components to the SoC

enjoy-digital, 2020-12-30:
Some CPUs (Microwatt/VexRiscv-SMP for example) requires external components to be added to the SoC (PLIC/CLINT for example with VexRiscv-SMP). For now this has been done with specific workarounds but it would be nice to add a generic method on the CPUs that would be called automatically if present after adding the CPU and that would add the components to the SoC.

enjoy-digital, 2020-12-30:
Implemented with https://github.com/enjoy-digital/litex/commit/74844db3b9fdc663aee61b2a1f3730d38664d04e. This allows some simplifications in Linux-on-LiteX-VexRiscv: https://github.com/litex-hub/linux-on-litex-vexriscv/commit/ad398759b5aba7cc2c12aac5355cf3600599b7ff.

closed 2020-12-30 by enjoy-digital

mithro, 2020-12-30:
@enjoy-digital - This is a great feature! Thanks for adding it.