enjoy-digital/litex #755: Wishbone GDB connection not possible to Arty A7+VexRISCV

dayjaby, 2021-01-01:
I created a bitstream for Arty A7 via the command:

```
python3 ./arty.py --toolchain vivado --cpu-type vexriscv --sys-clk-freq 80e6 --with-etherbone --csr-csv csr.csv --build
```

where I set the IP address of the board to 192.168.178.250:

```
if with_etherbone:
    self.add_etherbone(phy=self.ethphy, ip_address="192.168.178.250")
```

When I try to establish a connection to it, I get an error with the wishbone-tool version 0.7.8 and version 0.7.7:

```
wishbone-tool --ethernet-host 192.168.178.250 -s gdb
thread '<unnamed>' panicked at 'couldn't start server: RiscvCpuError(InstructionTimeout)', crates/core/main.rs:381:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

When merely reading some memory locations it works however:

```
$ wishbone-tool --ethernet-host 192.168.178.250 0x40000000
Value at 40000000: 00c000ef
$ wishbone-tool --ethernet-host 192.168.178.250 0x40000010
Value at 40000010: 61c10113
```

After specifying the CSR csv file, I get another error message:

```
$ wishbone-tool --ethernet-host 192.168.178.250 --csr-csv litex/boards/targets/csr.csv -s gdb
Error: "invalid configuration: GDB specified but no vexriscv address present in csv file"
```

Is the vexriscv address the same address that you would set for the cpu debug_bus? E.g.
```
        self.register_mem("vexriscv_debug", 0xf00f0000, self.cpu.debug_bus, 0x100)
```

which resulted in

```
AttributeError: 'VexRiscv' object has no attribute 'debug_bus'
```

After adding add_debug, I get another bunch of errors:

```
        self.cpu.add_debug()
        self.register_mem("vexriscv_debug", 0xf00f0000, self.cpu.debug_bus, 0x100)
```

the errors:
```
ERROR: [Synth 8-448] named port connection 'debugReset' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20611]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_address' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20612]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_data' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20613]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_payload_wr' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20614]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_valid' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20615]
ERROR: [Synth 8-448] named port connection 'debug_bus_cmd_ready' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20632]
ERROR: [Synth 8-448] named port connection 'debug_bus_rsp_data' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20633]
ERROR: [Synth 8-448] named port connection 'debug_resetOut' does not exist for instance 'VexRiscv' of module 'VexRiscv' [/home/david/Development/litex/litex/litex/boards/targets/build/arty/gateware/arty.v:20634]
```

What else am I missing here?

xobs, 2021-01-02:
You need to specify a CPU variant.

CPU cores with debug actually count as different variants. These debug cores expose additional wires, which you see expressed here as `debugReset`, `debug_bus_cmd_payload_address`, and so on.

To do so, you pass `variant="XXX"` to your CPU constructor. A list of valid variants is available at https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/core.py#L23-L39

For debug, you want to specify a core that has `+debug` at the end. For example, `variant="imac+debug"`.

There maybe should be a check to make sure this situation doesn't happen, where you try to instantiate a core with debug wires when debug isn't specified.

dayjaby, 2021-01-02:
You are the best!

```
wishbone-tool --ethernet-host 192.168.178.250 --csr-csv litex/boards/targets/csr.csv -s gdb
INFO [wishbone_tool::server] accepting gdb connections on 127.0.0.1:3333
```

> There maybe should be a check to make sure this situation doesn't happen, where you try to instantiate a core with debug wires when debug isn't specified.

I leave this issue open and we can try to find a nice spot to place an early error message?

hansfbaier, 2021-01-05:
@xobs Thanks that also helped me. This should go into the documentation somewhere.

enjoy-digital, 2021-09-08:
Closing since has been answered. This has been labelled as add-answer-to-wiki and answer will be added to Wiki.

closed 2021-09-08 by enjoy-digital