enjoy-digital/litex #764: Issues implementing GPIOTristate

graygreenblue, 2021-01-05:
I'm very new to FPGA designs and the litex tools and I'm sure I'm missing something obvious, sorry if this is the wrong place to post this.

I'm trying to implement a Vexriscv CPU on an Arty A7 with tristate GPIO pins. I've taken the default Arty target and added the following two lines to the end of the BaseSoC class:

`self.submodules.gpio = GPIOTristate(platform.request_all("user_led"))`
`self.add_csr("gpio")`

When I run the build opperation with the following command:
`python3 litex/litex-boards/litex_boards/targets/arty.py --build`

I get the following synthesis errors from vivado:

> Finished Renaming Generated Instances : Time (s): cpu = 00:01:08 ; elapsed = 00:01:12 . Memory (MB): peak = 2473.309 ; gain = 155.156 ; free physical = 807 ; free virtual = 6709
> 
> ---------------------------------------------------------------------------------
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple0_i_inferred__0_n_0 with 1st driver pin 'tstriple0_i_inferred_i_1/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple0_i_inferred__0_n_0 with 2nd driver pin 'user_led0_IBUF_inst/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple1_i_inferred__0_n_0 with 1st driver pin 'tstriple1_i_inferred_i_1/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple1_i_inferred__0_n_0 with 2nd driver pin 'user_led1_IBUF_inst/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple2_i_inferred__0_n_0 with 1st driver pin 'tstriple2_i_inferred_i_1/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple2_i_inferred__0_n_0 with 2nd driver pin 'user_led2_IBUF_inst/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple3_i_inferred__0_n_0 with 1st driver pin 'tstriple3_i_inferred_i_1/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]
> CRITICAL WARNING: [Synth 8-6859] multi-driven net on pin tstriple3_i_inferred__0_n_0 with 2nd driver pin 'user_led3_IBUF_inst/O' [/home/user/Desktop/litex/build/arty/gateware/arty.v:4]

And these errors from DRC:

> Starting DRC Task
> INFO: [DRC 23-27] Running DRC with 4 threads
> ERROR: [DRC MDRV-1] Multiple Driver Nets: Net tstriple0_i has multiple drivers: tstriple0_i_inferred_i_1/O, and user_led0_IBUF_inst/O.
> ERROR: [DRC MDRV-1] Multiple Driver Nets: Net tstriple1_i has multiple drivers: tstriple1_i_inferred_i_1/O, and user_led1_IBUF_inst/O.
> ERROR: [DRC MDRV-1] Multiple Driver Nets: Net tstriple2_i has multiple drivers: tstriple2_i_inferred_i_1/O, and user_led2_IBUF_inst/O.
> ERROR: [DRC MDRV-1] Multiple Driver Nets: Net tstriple3_i has multiple drivers: tstriple3_i_inferred_i_1/O, and user_led3_IBUF_inst/O.
> ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

If anyone can point me in the right direction I'd really appreciate it.

hansfbaier, 2021-01-05:
Do you have the LED chaser still activated?
In litex_boards/targets/arty.py for instance you find:
```
        self.submodules.leds = LedChaser(
            pads         = platform.request_all("user_led"),
            sys_clk_freq = sys_clk_freq)
        self.add_csr("leds")
```
This also drives the LED ports thus creating a driver-driver conflict, which vivado complained about.

enjoy-digital, 2021-01-06:
Thanks for the feedback @graygreenblue.  Unfortunately due to the bi-directional nature a tristate, `Platform.request_all` will not be supported and incorrect verilog was generated, which was then reported by Vivado. I added an assertion to detect this issue earlier directly during elaboration with https://github.com/enjoy-digital/litex/commit/0984308318a2bdb99a9662fd37a82c995376b2f9.

So in your case you can use:

```
self.submodules.gpio0 = GPIOTristate(platform.request("user_led", 0))
self.add_csr("gpio0")
self.submodules.gpio1 = GPIOTristate(platform.request("user_led", 1))
self.add_csr("gpio1")
```

or if you want to have multi-bit GPIO, define the IOs in your platform file:

`("user_leds", 0, Pins("H5 J5 T9 T10"),  IOStandard("LVCMOS33"))`

And use it in your target:

```
self.submodules.gpio = GPIOTristate(platform.request("user_leds"))
self.add_csr("gpio")
```

graygreenblue, 2021-01-06:
Thanks enjoy-digital, unfortunately I've tried both of these suggestions and I still get the same errors. I'm more interested in the multi-bit GPIO but I'm suspicious I'm having the same problem with both. 

I've tried making some changes to how the `GPIOTristate` is instantiated including looking back at previous commits for it with no success. The generated verilog, most seems like pretty straight forward assignments, is it possible the issue is caused by `xilinxmultiregimpl1_regs1`?

enjoy-digital, 2021-01-06:
@graygreenblue: ah sorry, you are right. This should be fine now with https://github.com/enjoy-digital/litex/commit/5f9c4a4ab4e8e959ef9cf38291d0e60e25571270. (I tested the full synthesis/p&r this time).

graygreenblue, 2021-01-06:
@enjoy-digital Thanks for all the help, it works great now!

closed 2021-01-06 by graygreenblue