enjoy-digital/litex #765: Modularize generated verilog: Separate modules into files and put wiring in top.v

hansfbaier, 2021-01-05:
Currently, litex generateds some monolithic 20k loc verilog file,
which is humanly next to incomprehensible.
It would help greatly to factor out all hardware modules into separate files
each and put the wiring only in the top.v module.

enjoy-digital, 2021-01-05:
Hi @hansfbaier,

this is indeed that has already been identified and something we'd like to improve in the long term. Sadly removing this limitation requires non-trivial changes to Migen or a switch for LiteX to another HDL DSL. We are considering evaluating nMigen or have our own HDL DSL for that: https://github.com/enjoy-digital/litex/wiki/Migen-nMigen.

I'd like to only keep track in LiteX's issue of realistic short/mid term issues/features so I'm moving this to https://github.com/litex-hub/ideas/issues that lists more general ideas/wishes for the project.

hansfbaier, 2021-01-05:
@enjoy-digital  Oh great I was not aware of that wiki page. Thanks!

enjoy-digital, 2021-01-05:
@hansfbaier: sorry, it seems I'm not able to transfer the issue there. I'll open another issue in LiteX soon with the long term plans and we'll make sure to include your suggestions for https://github.com/enjoy-digital/litex/issues/765 and https://github.com/enjoy-digital/litex/issues/766 in it.

enjoy-digital, 2021-01-06:
@hansfbaier: I'm moving this to https://github.com/enjoy-digital/litex/issues/769 that is an issue we can now use to discuss ideas for the project without creating specific issues for them initially.

closed 2021-01-06 by enjoy-digital