enjoy-digital/litex #770: Need wiki page for: How to fix timing closure failures in litex SoC designs

hansfbaier, 2021-01-07:
Example: When I run the base acorn cle-215 design, I already get negative slack in the PCIe clock path,
but the hardware nevertheless works (only the DMA example shows a large number of errors).
As soon as I add the PCIe bridge for debugging, the timings get worse and the system falls apart
(only 0xff is returned from everything read from the FPGA, although the device still enumerates).
I really would need some information on how to fix timing issues (especially track down the failing signal paths
and how to map them to the python code, to find out which combinatorial elements need to be broken up.

enjoy-digital, 2021-01-07:
@hansfbaier: Some recent changes have probably introduced some timing errors in the Acorn-CLE 215 design that was previously building correctly with correct timings. I'll have  a look at this and will fix the behavior/timings. We could indeed list in the Wiki some methodology to check/apply timing constraints correctly with LiteX for basic things for users not familiar with FPGAs. But this will only be up to a certain extent, for more complex things we assume the tool is used by a FPGA designer with a  good understanding of FPGAs that is able to understand/apply and fix timing issues.

enjoy-digital, 2021-09-08:
The timing issues were related to missing false path constraints between the PCIe clock domain and System clock domain. It could indeed be nice in the future to add a Wiki page to explain the best practices to constraint a FPGA design, but for now this is a bit out of scope for the project and we don't currently have the time/resources to write this Wiki page. I'm closing this issue but adds the `add-answer-to-wiki` label to be able to easily find it in the future.

closed 2021-09-08 by enjoy-digital