enjoy-digital/litex #779: Serial core in separate clock domain should have its async FIFOs reset in system clock domain

cr1901, 2021-01-22:
Not sure the best way to proceed here is. Duplicating description of problem from IRC for convenience:

>(2:13:10 AM) cr1901_modern: _florent_: I figured out the problem I was seeing. It's relevant to orangecrab but may exist on other boards.
(2:14:53 AM) cr1901_modern: Basically, the USB clock domain on orangecrab wasn't being reset properly. This means the CDC gray code multiregs between USB and sys would not be updated with valid reset values, and the comparison to detect an empty receive queue from USB would be garbage
(2:16:00 AM) cr1901_modern: The CPU would always see a high level after reset from the RX interrupt. Since it's edge-triggered, the event never triggered, so the RX queue was never serviced
(2:17:31 AM) cr1901_modern: eventually, the RX queue fills up, and the USB core dies waiting for the RX queue to drain. Then when you send data from the CPU side, the TX queue fills up b/c the USB core refuses to drain it
(2:18:05 AM) cr1901_modern: Eventually, the BIOS dies in an infinite loop waiting for the TX queue to drain in a vicious cycle.

enjoy-digital, 2021-01-22:
Thanks for reporting and investigating on this. Would you mind testing with https://github.com/enjoy-digital/litex/commit/01a2fc11e2734477da9ec00b3eac5faa8231dbfb and https://github.com/litex-hub/litex-boards/commit/23760e2eae5fa819d8076a5440bada1ec7cc8000?

cr1901, 2021-01-22:
@enjoy-digital These two commits fix the issue I was seeing.

enjoy-digital, 2021-01-23:
Thanks for the test/feedback @cr1901.

closed 2021-01-23 by enjoy-digital