enjoy-digital/litex #786: Update CPU wiki page to match supported CPUs

mithro, 2021-01-27:
The directory at https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu has;
 - blackparrot
 - cv32e40p
 - lm32
 - microwatt
 - minerva
 - mor1kx
 - picorv32
 - rocket
 - serv
 - vexriscv
 - vexriscv_smp
 - zynq7000

The wiki page at https://github.com/enjoy-digital/litex/wiki/CPUs only has;
 - lm32 -- a LatticeMico32 soft core.
 - or1k -- an OpenRISC 1000 soft core (see also Open RISC on Wikipedia).
 - picorv32 -- a Small RISC V core by Clifford Wolf, implementing the rv32imc instruction set (or configured subsets)
 - vexriscv -- an FPGA Friendly RISC V core by SpinalHDL, implementing the rv32im instruction set (hardware multiply optional)
 - minerva -- a CPU core that currently implements the RISC-V RV32I instruction set with its microarchitecture described in plain  Python code using the nMigen toolbox.
 - rocket -- Rocket Chip, a configurable, fully featured, 64-bit rv64imafdc capable core.

Would be good to update the page a little bit.

mithro, 2021-01-27:
@kgugala -- As we have added a lot of these CPUs, we should probably do this work.

mithro, 2021-01-27:
I wonder if we should be generating the CPUs page from documentation directly inside the CPU cores Python files?

mithro, 2021-01-27:
@enjoy-digital -- What do you think of us putting the documentation the CPU core Python files and then generating a summary page published to the wiki?

enjoy-digital, 2021-01-27:
@mithro: Having the documentation in the CPU core python files could be interesting yes but the generating process should be simple (for example a simple script we can run that generates the wiki data to be copy paste). I'd like to avoid using a bot for that since this makes things complicated to maintain.

scanakci, 2021-02-21:
FYI, I updated the page to add BlackParrot.