enjoy-digital/litex #798: Arty A7: litex_term serial boot not working with vexriscv_smp

dayjaby, 2021-02-02:
After running the following lines I could not boot the Arty A7:

```
python3 ./arty.py --toolchain vivado --cpu-type vexriscv_smp --sys-clk-freq 100e6 --with-etherbone --csr-csv csr.csv --load
python3 litex_term.py --serial-boot /dev/ttyUSB1
```

Replacing vexriscv_smp with vexriscv makes it work. With GDB I also tested the first commands starting at 0x000000:
```
(gdb) x/5i 0x0
   0x0:	unimp
   0x2:	0xf00f
   0x6:	0x8013f00f
   0xa:	unimp
   0xc:	unimp
```

With vexriscv I get a "j, nop, nop, nop, ..." at this point, just as the objdump for the bios.elf suggests and how it is defined in the crt0.S:
```
00000000 <_ftext>:
       0:       a8ad                    j       7a <crt_init>
       2:       0001                    nop
       4:       0001                    nop
       6:       0001                    nop
       8:       0001                    nop
       a:       0001                    nop
       c:       0001                    nop
       e:       0001                    nop
```

Is something causing the bios to not be uploaded correctly for vexriscv_smp? Is the uart wishbone interface not working properly here?

enjoy-digital, 2021-02-03:
@dayjaby: Strange, are you sure the BIOS is been re-compiled correctly when switching from vexriscv to vexriscv-smp (rm -rf build)?

dayjaby, 2021-02-04:
Oh, you are right. Gonna make a PR so that the BIOS will automatically get recompiled. Closing this.

closed 2021-02-04 by dayjaby