enjoy-digital/litex #814: The software demo does not build

nickoe, 2021-02-11:
I am trying to build the demo project in the software dir as described in the README but it fails:

```Interactive
root@9517a373bc4c:/litex# git log -1 --oneline
041aa9bf (HEAD -> master, origin/master, origin/HEAD) soc/cores/clock/xilinx_us/USIDELAYCTRL: make sure sys clock domain is reseted when reference clock domain is reseted.
root@9517a373bc4c:/litex# litex_bare_metal_demo --build-path=build/arty/
Makefile:4: /software/common.mak: No such file or directory
make: *** No rule to make target '/software/common.mak'.  Stop.
cp: cannot stat 'demo/demo.bin': No such file or directory
```

I tested this in a plain docker image based on ubuntu to be sure I didn't have anything on my own system that I messed up. 

Essentially the dockerfile looks like this and above was just run interactively to show the errors more reabable.

```Dockerfile
FROM ubuntu:20.04

RUN apt-get update && apt-get install wget python3 -y
RUN apt-get update && apt-get install git build-essential -y
RUN apt-get update && apt-get install python3-setuptools -y
RUN wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
RUN chmod +x litex_setup.py
RUN ./litex_setup.py init install

RUN ./litex_setup.py gcc

RUN which litex_bare_metal_demo
RUN litex_bare_metal_demo --build-path=build/arty/
```

tcal-x, 2021-02-11:
It looks like $SOC_DIRECTORY is not being set, but I don't know where it usually gets set.    When I run from the shell, I don't need to set it manually.

tcal-x, 2021-02-11:
Oh, if you look at the demo Makefile: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/demo/Makefile

it has this include right before it uses $SOC_DIRECTORY,
```
include $(BUILD_DIR)/software/include/generated/variables.mak

```
and SOC_DIRECTORY should be getting defined in there (it is for me).  So try dumping that file to debug what is going on with your docker build.

nickoe, 2021-02-11:
Ok, florent gave a hint. It depends on a project being prepared, so essentially one could do:
```
litex-boards/litex_boards/targets/arty.py --no-compile-gateware --output-dir=/build/arty
litex_bare_metal_demo --build-path=/build/arty/
```

nickoe, 2021-02-11:
But for my normal env the installed script does not seem to work, although calling the demp.py direcly does:
```
[nickoe@x280-arch targets]$ litex_bare_metal_demo --build-path=build/mars_ax3/
Traceback (most recent call last):
  File "/home/nickoe/.local/bin/litex_bare_metal_demo", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'litex_bare_metal_demo')())
  File "/home/nickoe/.local/bin/litex_bare_metal_demo", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 79, in load
    return functools.reduce(getattr, attrs, module)
AttributeError: module 'litex.soc.software.demo.demo' has no attribute 'main'
```
And directly:
```
[nickoe@x280-arch targets]$ python ../../../litex/litex/soc/software/demo/demo.py --build-path=build/mars_ax3/
 CC       main.o
 LD       demo.elf
chmod -x demo.elf
 OBJCOPY  demo.bin
chmod -x demo.bin
```

PR #815 (2021-02-12): Change wording of demo README
Change wording of demo README to make it more clear what the process is
and how things related.  This should help the newcomer and it still
usefull for the triained.

Change the command example to be more copy paste friendly.

Fixes #814
commit e7e28f24 (2021-02-12): Change wording of demo README

Change wording of demo README to make it more clear what the process is
and how things related.  This should help the newcomer and it still
usefull for the triained.

Change the command example to be more copy paste friendly.

Fixes #814
closed 2021-02-12 by enjoy-digital