enjoy-digital/litex #831: Verilator trace is never written to disk properly

techninja1008, 2021-02-25:
I've been trying to get my design to simulate properly with Verilator using `SimPlatform`. I've discovered, however, that the outputted VCD file is at best malformed (with GTKWave refusing to open it) and often completely empty.

Browsing through the harness code shows that neither of `tfp->close()` or `tfp->flush()` are present, meaning that the trace file isn't written even when the simulation finishes naturally. I've found that adding `VerilatedVcd::flush_all();` at the end of `litex_sim_dump()` in sim_init.cpp and manually building and running the simulation results in a working trace file.

nickoe, 2021-02-26:
@techninja1008  Could this be, maybe tangentially, related to https://github.com/enjoy-digital/litex/issues/712 ?

techninja1008, 2021-03-03:
As far as I can tell, that issue does not seem to be related to this one. In this case, `dump()` is called correctly and successfully however the file is never flushed to disk.

nickoe, 2021-04-30:
@techninja1008 Can you reproduce this with litex_sim.py? If so, what commandline args do you use? I think we need an easy way to reproduce your steps to make progress on this.

developandplay, 2021-06-10:
I also encountered this issue a few times (seems to be flaky). After applying `vcd2fst` I could view the malformed VCD files.
Next time it occurs I will share the arguments.

GuzTech, 2021-09-14:
I have the same issue as the OP. To reproduce, simply run litex_sim.py using upstream LiteX. In my case, the contents of the VCD file just stops abruptly, indicating the non-existence of a flush or something similar.

krabo0om, 2023-04-07:
I'm facing the same issue, the VCD is void of any updates, only the signal definitions. Is there any fix or known work around?