enjoy-digital/litex #834: CSRStatus may be rw but is exported always as ro

mczerski, 2021-02-28:
https://github.com/enjoy-digital/litex/blob/6e883b4513232e9cf8b017f7aaa08b3f2de7b062/litex/soc/integration/export.py#L288

the CSRStatus may be created with read_only=False, which means it is read/write so during exporting to csv/json memory map it should be taken into account.

This happens when instantiating GPIOIn module with argument "irqs" (to make gpio interupt).

Also i have noticed that I cannot clear the interrupt pending flag by writing to the pending register, is it possible that this is related to the generated memory map ?

mczerski, 2021-02-28:
>Also i have noticed that I cannot clear the interrupt pending flag by writing to the pending register, is it possible that this is related to the generated memory map ?

this part is false, it was just a dangling pin which was generating interrupts all the time.

enjoy-digital, 2021-08-24:
Thanks @mczerski, sorry for the delay, read_only mode has been added recently and was indeed not exported corretly, this is fixed with https://github.com/enjoy-digital/litex/commit/f017b06926643b71b1faca1cf607c9c4378026cd.

closed 2021-08-24 by enjoy-digital