enjoy-digital/litex #835: lxsim --cpu-type=vexriscv command not found in MacOS X

jhu960213, 2021-03-01:
Hello,

I'm just following the installation guide and everything seems to be working up until the last point where I want to simulate in verilator. All the packages have been installed with brew and it works but when I type this command in I get "zsh: command not found: lxsim". Is this a package issue or I'm not including something in my path for the command etc? Thanks! 

I'm running MacOS X 10.15.7

Best,
J.H

enjoy-digital, 2021-03-02:
Hello @jhu960213,

lxsim is part of the console scripts installed with LiteX: https://github.com/enjoy-digital/litex/blob/master/setup.py#L36-L51

Until we figure out why it's not available after the installation on your machine, you can find the script in litex.tools.litex_sim.

jhu960213, 2021-03-02:
Hello @enjoy-digital ,

I actually was able to find lxsim as the binary itself was in Library/Python/3.8/bin/ folder and I executed the binary with the full path. During the build phase, it ran into some errors. Attached below are my outputs of the error. 
[image]
[image]

Let me know if this is enough for troubleshooting what is going on? Thanks! 

Best,
J.H

enjoy-digital, 2021-03-08:
@jhu960213: Your issue has probably been fixed with https://github.com/enjoy-digital/litex/commit/d3407c67b189d2c338766febb3bf2c57f6f80d1d, could you update and try?

marrrk, 2021-05-10:
> Hello @jhu960213,
> 
> lxsim is part of the console scripts installed with LiteX: https://github.com/enjoy-digital/litex/blob/master/setup.py#L36-L51
> 
> Until we figure out why it's not available after the installation on your machine, you can find the script in litex.tools.litex_sim.

Hi, I received the same error (command not found) after following all the installation steps. I'm not running Mac OS however, I'm running POP OS 20.04. Using this comment I was able to run the simulation just fine.

enjoy-digital, 2021-09-08:
Sorry the issue seems specific to your installation, other MacOS X user have been able to use LiteX on their machine. You probably resolved the issue so I'm closing but feel free to re-open if that's not the case, we'll try to have help from a MacOS user.

closed 2021-09-08 by enjoy-digital