enjoy-digital/litex #839: Problems reading CSRs on ZynqUS+ via PS AXI

jedrzejboczar, 2021-03-04:
In the setup described here: https://github.com/antmicro/litex-rowhammer-tester/issues/22 we had issues with reading UART over EtherBone. The output was malformed, for example:
```
[m Bidyu adae aiym

 c oyih 0222 no-iia
BO ul nMr  011:33
LtXgtsa:fd442)

-========SC[m=========-
[mS:    2btdt-i  GB
[mRM[m  8iB
[mDA:   087KB6-i  00Ts(L9CL9

-===== [mntaiain[m======-
rt eeig osfwr oto.
  | | ||bs:-
 0 1dly kdlyt 1tp.
 m:|dly
 m:|dly
 m:|dly
 m:|dly
 m:|dly
 m:|0dly
 m:|dly
Ra eeig
 0 1 0011111111000000|dly:16-3
 m,b:|0000000000111111 eas 2+9
 0 5 0000000000000000|dly:
 m,b:|0000000000000000 eas
 1 0 1000000000000000|dly:0+0
 1 5 0000000000000000|dly:-0-19
 1 7 0000000000000000|dly:
 bs:m,b1dly:18-2
 m,b:|0000000000000000 eas
 2 3 0000000000000000|dly:0+10
 m,b:|000000000000000 eas
 m,b:|00000000000000 eas
 m,b:|0000000000000000 eas
 3 3 0000000000000000|dly:0+18
 m,b:|00000000000000 eas
 3 6 0000000000000000 eas
 4 1 1111111000000000|dly:9+9
 4 4 0000000000000000|dly:-5+13
 m,b:|0000000000000|dly:
 m,b:|0000000000000000 eas
 m,b:|00000000000000 eas
 5 0 0000000000000000|dly:
 5 3 0000000000000000|dly:4+18
 m,b:|0000000000000000 eas
 m,b:|0000000000000000 eas
 m,b:|000000000000000 eas
 m,b:|00000000000000 eas
 6 3 0000000000000001|dly:9+24
 m,b:|0000000000000000 eas
 6 6 0000000000000000|dly:
 m,b:|0000000000000000 eas
 7 1 1111100000000000|dly:6+6
 7 7 0000000000000000|dly:-2-02
 bs:m,b2dly:27-3
SicigSRMt adaecnrl
 rt:04000-x00002i   Wie x000004100 MB B
 Ra:04000-x0e001i   Ra:04000-x00002i
MmetO
Mmpe t04000 2i).
```

We were able to work around it by using 32-bit PS<->PL AXI, but it is not a fix for the original issue. 

Reading the UART output was done similarily as in litex_crossover_uart, the exact code is here: https://github.com/antmicro/litex-rowhammer-tester/blob/acf9f8de44fb5ac2fb792dfb18a3f72969a38fb3/rowhammer_tester/scripts/mem.py#L56-L61.
PC was connected to ZCU104 to PS Ethernet. PS was running simple `zcu104_etherbone` server. On a read request it was accessing memory mapped to AXI connected to PL:
* callback: https://github.com/antmicro/litex-rowhammer-tester/blob/acf9f8de44fb5ac2fb792dfb18a3f72969a38fb3/firmware/zcu104/etherbone/etherbone.c#L109
* actual read from mmap memory: https://github.com/antmicro/litex-rowhammer-tester/blob/acf9f8de44fb5ac2fb792dfb18a3f72969a38fb3/firmware/zcu104/etherbone/main.c#L19-L25

On PL side the conversion was done the following way: https://github.com/antmicro/litex-rowhammer-tester/blob/acf9f8de44fb5ac2fb792dfb18a3f72969a38fb3/rowhammer_tester/targets/zcu104.py#L175-L203.

After including your fix (https://github.com/enjoy-digital/litex/commit/fc83a9281ae15f936af6cd0abd19d4d48f57b895) the problem is still there, so it seems that it wasn't the only issue.

To test what is happening I added LiteScope in the ZCU104 target (didn't add data as it is wide but may also try with it if needed):
```python
# Analyzer ---------------------------------------------------------------------------------
analyzer_signals = [
    axi_ps.ar,
    axi_ps.r.valid,
    axi_ps.r.ready,
    axi_ps.r.resp,
    axi_ps.r.id,
    axi_ps.r.last,
]
from litescope import LiteScopeAnalyzer
self.submodules.analyzer = LiteScopeAnalyzer(analyzer_signals,
   depth        = 512,
   clock_domain = "sys",
   csr_csv      = "analyzer.csv")
self.add_csr("analyzer")
```
I had issues trying to use LiteScope over EtherBone, so I added UartBone via the UART connected from PL to the onboard FTDI and used it to capture a dump (one litex_server for litescope over UartBone, one litex_server for the CSR access over PS EtherBone). I attach example litescope dumps of performing a single read on UART crossover CSRs:
[dumps.zip](https://github.com/enjoy-digital/litex/files/6084630/dumps.zip). They seem to be correct 32-bit transfers, burst=incr, size=4bytes, len=1transfer.

Please ask if you need any more information or want me to perform some additional tests.

enjoy-digital, 2021-03-10:
@jedrzejboczar: I just looked at it and created a minimal simulation to investigate (attached if useful later).

In this test I'm just trying to be in a situation close to yours. While looking at the simulation traces, I noticed that the base address of downconverter accesses is not good: It was starting on aligned's master addresses: A 128-bit Master reading at 0x00000808 was in fact doing 32-bit accesses on  0x00000800, 0x00000804, 0x00000808, 0x0000080c instead of 0x00000808, 0x0000080c, 0x00000810, 0x00000814).

PR https://github.com/enjoy-digital/litex/pull/846 should  fix it (It at least fixed my simulation), can you try with it?

```python3
#!/usr/bin/env python3

import os
import argparse

from migen import *

from litex.build.generic_platform import *
from litex.build.sim import SimPlatform
from litex.build.sim.config import SimConfig

from litex.soc.integration.soc_core import *
from litex.soc.integration.builder import *

# Platform -----------------------------------------------------------------------------------------

_io = [
    ("sys_clk", 0, Pins(1)),
    ("sys_rst", 0, Pins(1)),
]

class Platform(SimPlatform):
    default_clk_name = "sys_clk"
    def __init__(self):
        SimPlatform.__init__(self, "SIM", _io)

# SimAXILite128b -----------------------------------------------------------------------------------

class SimAXILite128b(SoCMini):
    def __init__(self, sys_clk_freq=int(1e6)):
        # Platform ---------------------------------------------------------------------------------
        platform  = Platform()
        self.comb += platform.trace.eq(1) # Always enable tracing.

        # SoCMini ----------------------------------------------------------------------------------
        SoCMini.__init__(self, platform, clk_freq=sys_clk_freq)

        # UART -------------------------------------------------------------------------------------
        from litex.soc.cores.uart import UART
        self.submodules.uart = UART(rx_fifo_rx_we=True)
        self.add_csr("uart")

        # Send Counter on UART's Sink.
        count = Signal(8)
        self.sync += [
            self.uart.sink.valid.eq(1),
            self.uart.sink.data.eq(count),
            If(self.uart.sink.ready,
                count.eq(count + 1)
            )
        ]

        # AXI --------------------------------------------------------------------------------------
        from litex.soc.interconnect.axi import AXILiteInterface
        axi_lite_128b = AXILiteInterface(data_width=128)
        self.bus.add_master(master=axi_lite_128b)

        # Test -------------------------------------------------------------------------------------
        UART_RXTX_ADDR    = 0X00000800
        UART_RXEMPTY_ADDR = 0X00000808

        display = Signal()
        self.submodules.fsm = fsm = FSM(reset_state="READ-RX-EMPTY-CMD")
        fsm.act("READ-RX-EMPTY-CMD",
            axi_lite_128b.ar.valid.eq(1),
            axi_lite_128b.ar.addr.eq(UART_RXEMPTY_ADDR),
            If(axi_lite_128b.ar.ready,
                NextState("READ-RX-EMPTY-DATA")
            )
        )
        fsm.act("READ-RX-EMPTY-DATA",
            axi_lite_128b.r.ready.eq(1),
            If(axi_lite_128b.r.valid,
                NextState("READ-RX-CMD")
            )
        )
        fsm.act("READ-RX-CMD",
            axi_lite_128b.ar.valid.eq(1),
            axi_lite_128b.ar.addr.eq(UART_RXTX_ADDR),
            If(axi_lite_128b.ar.ready,
                NextState("READ-RX-DATA")
            )
        )
        fsm.act("READ-RX-DATA",
            axi_lite_128b.r.ready.eq(1),
            If(axi_lite_128b.r.valid,
                display.eq(1),
                NextState("READ-RX-EMPTY-CMD")
            )
        )
        self.sync += If(display, Display("DATA: %32x", axi_lite_128b.r.data))
        count = Signal(32)
        self.sync += count.eq(count + 1)
        self.sync += If(count == 10000, Finish())

# Build --------------------------------------------------------------------------------------------

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("--trace",       action="store_true", help="Enable VCD tracing")
    args = parser.parse_args()
    sim_config = SimConfig()
    sim_config.add_clocker("sys_clk", freq_hz=1e6)

    soc = SimAXILite128b()
    builder = Builder(soc, output_dir="build", csr_csv="csr.csv")
    builder.build(sim_config=sim_config, trace=args.trace)

if __name__ == "__main__":
    main()
```

enjoy-digital, 2021-03-10:
@jedrzejboczar: Sorry the CI is not passing, I'll have a closer look.

enjoy-digital, 2021-03-10:
@jedrzejboczar: This is good now.

jedrzejboczar, 2021-03-18:
Sorry for the delay. I was able to test it today with you fix and it doesn't seem to work entirely correctly, however there is some improvement. 

When I try to read the crossover uart the correct way, checking `uart_xover_rxempty`, then it is never empty. However if I remove the check, then just reading `uart_xover_rxtx` gives all the UART output we would expect, but sometimes the same character is read more than once, which is expected when not checking `uart_xover_rxempty`.

Dumping the registers yields:
```
0x82000000: 0x00000000 ctrl_reset
0x82000004: 0x00000000 ctrl_scratch
0x82000008: 0x00000000 ctrl_bus_errors
0x82002000: 0x00000000 uart_rxtx
0x82002004: 0x00000001 uart_txfull
0x82002008: 0x00000000 uart_rxempty
0x8200200c: 0x00000000 uart_ev_status
0x82002010: 0x00000000 uart_ev_pending
0x82002014: 0x00000000 uart_ev_enable
0x82002018: 0x00000070 uart_txempty
0x8200201c: 0x00000000 uart_rxfull
0x82002020: 0x00000065 uart_xover_rxtx
0x82002024: 0x00000000 uart_xover_txfull
0x82002028: 0x00000002 uart_xover_rxtx
0x8200202c: 0x00000001 uart_xover_ev_status
0x82002030: 0x00000002 uart_xover_ev_pending
0x82002034: 0x00000001 uart_xover_ev_enable
0x82002038: 0x00000000 uart_xover_txempty
0x8200203c: 0x00000000 uart_xover_rxfull
0x82002800: 0xffffffff timer0_load
0x82002804: 0x00000001 timer0_reload
0x82002808: 0xfdd19eca timer0_en
0x8200280c: 0x00000001 timer0_update_value
0x82002810: 0xfdd19eca timer0_value
(...)
```
So e.g. `ctrl_scratch`, `uart_xover_rxtx`, `timer0_en` have incorrect values, but `uart_xover_rxtx` seems to have a correct one, so it seems that it has something to do with its address alignment.

jedrzejboczar, 2021-03-18:
I did some dumps with LiteScope, this time more and with the read data also being captured: [dumps.zip](https://github.com/enjoy-digital/litex/files/6164974/dumps.zip)

I created a simple script which captured dumps for a list of registers and then repeated it once more. The files are named as `f'{reg}__addr_0x{r.addr:08x}__iter_{i}__data_0x{data:08x}.vcd'`, with `data` being the return value from `RemoteClient` read.

Before the capture I did a few reads from `uart_xover_rxtx` until start of the line `BIOS CRC passed`. However the values of subsequent captures of `uart_xover_rxtx` are 0x70 and 0x65 which means 'p' and 'e', so I guess some other reads have consumed the missing 'a' and two 's' characters.

enjoy-digital, 2021-03-18:
Thanks @jedrzejboczar, when looking at `ctrl_scratch__addr_0x82000004__iter_0__data_0x00000000.vcd`, we can see that the returned data on the AXI bus seems fine: `0x0000000000000000000000001234578`. That would be interesting to look at how this returned data is seen by the software running on the Zynq.

jedrzejboczar, 2021-03-18:
Hmm, indeed it looks good. I'll try to run the software etherbone server with debug output.

jedrzejboczar, 2021-03-18:
It seems that it just does a 32-bit read to the memory mapped memory and receives incorrect value. Here are 2 reads of ctrl_scratch then ctrl_reset:
```
Received 20 byte packet
4e 6f 10 44 00 00 00 00
00 0f 00 01 00 00 00 00
82 00 00 04
0x82000004 => 0x00000000
Sending 20 byte response
4e 6f 31 44 00 00 00 00
00 0f 01 00 00 00 00 00
00 00 00 00
Received 20 byte packet
4e 6f 10 44 00 00 00 00
00 0f 00 01 00 00 00 00
82 00 00 04
0x82000004 => 0x00000000
Sending 20 byte response
4e 6f 31 44 00 00 00 00
00 0f 01 00 00 00 00 00
00 00 00 00
Received 20 byte packet
4e 6f 10 44 00 00 00 00
00 0f 00 01 00 00 00 00
82 00 00 00
0x82000000 => 0x00000000
Sending 20 byte response
4e 6f 31 44 00 00 00 00
00 0f 01 00 00 00 00 00
00 00 00 00
```
This is the output from the following code accessing the memory in PS: https://github.com/antmicro/litex-rowhammer-tester/blob/acf9f8de44fb5ac2fb792dfb18a3f72969a38fb3/firmware/zcu104/etherbone/main.c#L22-L23.

enjoy-digital, 2021-03-18:
Thanks, it indeed seems the MMAP is not behaving correctly on un-aligned accesses. (I think the SoC is behaving correctly now but not the software).

jedrzejboczar, 2021-03-18:
Ok, thanks for help. It seems that everything works fine on the Litex side now, so I will close this issue. In https://github.com/antmicro/litex-rowhammer-tester we already reconfigured the PS<->PL AXI to 32-bit in FSBL avoiding the problem.

closed 2021-03-18 by jedrzejboczar

enjoy-digital, 2021-03-18:
Thanks @jedrzejboczar, yes that's maybe not necessary to investigate more if you are using 32-bit AXI now. I'll just merge the PR.