enjoy-digital/litex #844: create_clock fails in Vivado v2019.2

japm48, 2021-03-10:
I get this when building for Arty in Vivado v2019.2:
```
CRITICAL WARNING: [Vivado 12-4739] create_clock:No valid object(s) found for '-objects [get_nets clk100]'.
```

I didn't try to continue the build as it sounds too serious...

### Workaround

I managed to fix this by just changing `get_nets` to `get_ports`; that is, in the constraints file change this line:
```
create_clock -name clk100 -period 10.0 [get_nets clk100]
```
To this:
```
create_clock -name clk100 -period 10.0 [get_ports clk100]
```

I can write the trivial patch in [vivado.py](https://github.com/enjoy-digital/litex/blob/master/litex/build/xilinx/vivado.py) but I don't know if this breaks anything else.

japm48, 2021-03-12:
Ok, as expected changing from get_nets to get_ports breaks the generated clocks (from PLLs), so there should be a distinction between external clocks and generated clocks. A quick way to solve this would be to check if the clock name is in the input list. An alternative would be to create an internal signal.

I believe this is important as many warnings get removed, even if a BUFG apparently gets inferred, the clock groups command does not appear to work.

Also, there should be no need to mark the clocks as `dont_touch`, this also removes a warning (possibly harmless??): `Clock pin has keep related attribute (keep/mark_debug/dont_touch) which could create extra logic on its net`.

enjoy-digital, 2021-03-12:
Thanks @japm48, this is probably a regression we haven't noticed. I confirm I see the same critical warning and being able to know if the clock is an input or internal signal would indeed allow selecting between `get_nets` and g`et_ports`. I'll try to look at this soon.

enjoy-digital, 2021-03-12:
@japm48: Note even with the critical warning, the timing constraint still seems to be correctly applied (probably through the PLL's 
`CLKIN1_PERIOD`), but I'll fix the critical warning and will also try to remove the other warnings.

````
Clock                  Waveform(ns)         Period(ns)      Frequency(MHz)
-----                  ------------         ----------      --------------
clk100                 {0.000 5.000}        10.000          100.000         
  crg_clkout0          {0.000 5.000}        10.000          100.000         
  crg_clkout1          {0.000 1.250}        2.500           400.000         
  crg_clkout2          {0.625 1.875}        2.500           400.000         
  crg_clkout3          {0.000 2.500}        5.000           200.000         
  crg_clkout4          {0.000 20.000}       40.000          25.000          
  subfragments_pll_fb  {0.000 5.000}        10.000          100.000         
eth_rx_clk             {0.000 20.000}       40.000          25.000          
eth_tx_clk             {0.000 20.000}       40.000          25.000
````

japm48, 2021-03-13:
Thanks @enjoy-digital.

So, apparently it is not as critical as Vivado implies. Also, this only happens in synthesis and I guess that in implementation the net is already created so it can be found.

Anyways, after applying my suggested changes I still get this warning in both synthesis and implementation:

```
[Vivado 12-3521] Clock specified in more than one group: builder_s7mmcm_mmcm_fb, clkout0, clkout1, and main_crg_clkout0 [".../arty.xdc":513]
```

The relevant part of the xdc file:
```tcl
create_clock -name clk100 -period 10.0 [get_ports clk100]
create_clock -name eth_rx_clk -period 40.0 [get_nets eth_rx_clk]
create_clock -name eth_tx_clk -period 40.0 [get_nets eth_tx_clk]
set_clock_groups -group [get_clocks -include_generated_clocks -of [get_nets sys_clk]] -group [get_clocks -include_generated_clocks -of [get_nets main_crg_clkin]] -asynchronous
set_clock_groups -group [get_clocks -include_generated_clocks -of [get_nets sys_clk]] -group [get_clocks -include_generated_clocks -of [get_nets eth_rx_clk]] -asynchronous
set_clock_groups -group [get_clocks -include_generated_clocks -of [get_nets sys_clk]] -group [get_clocks -include_generated_clocks -of [get_nets eth_tx_clk]] -asynchronous
set_clock_groups -group [get_clocks -include_generated_clocks -of [get_nets eth_rx_clk]] -group [get_clocks -include_generated_clocks -of [get_nets eth_tx_clk]] -asynchronous
```

Maybe this is not the best place to ask, but what is the intended purpose of the clock groups? (at least for this use case)

enjoy-digital, 2022-04-22:
This has been fixed with https://github.com/enjoy-digital/litex/commit/6b3a5412416e89869073dbc32540e6f2282c70fc.

closed 2022-04-22 by enjoy-digital